Xilinx/XRT #535: DSA feature discovery based on VBNV name is hacky

maxzhen, 2018-10-19:
https://github.com/Xilinx/XRT/blob/96f4059b0e6fcbf54dce5fedd73217078b73ae2c/src/runtime_src/driver/xclng/drm/xocl/subdev/feature_rom.c#L317
It's better to add the feature info to either:
DSA's feature ROM
OR
static table defined in devices.h and associate it with the correct device ID.

stsoe, 2018-10-19:
Feature ROM would be ideal, but the info is not there.  I need to know, if a DSA (1) has ERT, (2) if ERT is there, does the CU address lookup through a table lookup, or is addressing consecutive, (3) does the DSA support CDMA.  

(3) will be addressed, I know, but I am not sure about (1) and (2).    I'd be happy to move the the information into the static tables in device.h if only I had a clue as how to determine which section correspond to which device and how I am supposed to add the information.

maxzhen, 2018-10-19:
Lizhi knows better about the details. But basically, you just need to add member(s) to struct xocl_board_private, which will be populated by a XOCL_BOARD_xxx macro, which then will be associated with a particular (vendor, device, subdevice) triplet in XOCL_MGMT_PCI_IDS or XOCL_USER_XDMA_PCI_IDS. Take a look at struct xocl_board_private->board_name for example.

stsoe, 2019-03-20:
I believe this has finally been taken care of.

closed 2019-03-20 by stsoe