Xilinx/XRT #544: The compiled documentation should not be in the source repository

keryell, 2018-10-19:
There should be no https://github.com/Xilinx/XRT/tree/master/docs directory in the Git repository, since it is a by-product of the documentation compilation.

I hope that this is not directly used by https://xilinx.github.io/XRT/ ...

keryell, 2018-11-06:
To make progress on this, one can look for example at https://github.com/triSYCL/triSYCL/blob/master/doc/architecture.rst#source-code-documentation about how to move the generated documentation into a separate branch.

mamin506, 2018-12-11:
@keryell Hi, Ronan. The docs/ folder was removed from master branch. If you run build.sh, the document will be built in Release/runtime_src/doc/ folder. Only Sonal and Soren can publish documents to gh-pages branch. See #765.

keryell, 2018-12-11:
That seems like a reasonable solution.
At least the git repository is quite cleaner now!
Thanks.

closed 2018-12-11 by keryell