Xilinx/XRT #627: No bitmap_to_u32array() in before Kernel 4.6

mamin506, 2018-11-06:
https://github.com/Xilinx/XRT/blob/21c61b080ee476bd03d1ee73cdf9f58d1105e5c4/src/runtime_src/driver/xclng/drm/xocl/subdev/mb_scheduler.c#L2028

Please check.
4.4.162  https://elixir.bootlin.com/linux/v4.4.162/source/include/linux/bitmap.h
4.5.7      https://elixir.bootlin.com/linux/v4.5.7/source/include/linux/bitmap.h
4.6         https://elixir.bootlin.com/linux/v4.6/source/include/linux/bitmap.h

I observed build failed will failed on CentOS 7.5 with Linux kernel 4.4.156.

@stsoe

stsoe, 2018-11-14:
u32array was an old API, it is moved to arr32.  The code switches to _arr32 in 4.16, looks like it has to switch earlier.  If you already know, can you please fix the macro.  Thanks!

stsoe, 2019-03-20:
We have worked around by implementing the needed functions in mb_scheduler.c itself

closed 2019-03-20 by stsoe