Xilinx/XRT #783: build.sh fail on PowerPC system

DJmimi35, 2018-12-14:
[build.sh_error.txt](https://github.com/Xilinx/XRT/files/2678308/build.sh_error.txt)
Hi,

I'm trying to install XRT on my PowerPC server, but it failed on build.sh step. The error message said:

/home/xilinx/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx: In member function ‘size_t xclhwemhal2::HwEmShim::xclReadSkipCopy(uint64_t, void*, size_t)’:
/home/xilinx/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx:1532:67: error: requested alignment 128 is larger than 16 [-Werror=attributes]
     alignas(DDR_BUFFER_ALIGNMENT) char buffer[DDR_BUFFER_ALIGNMENT];
                                                                   ^
/home/xilinx/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx: In member function ‘size_t xclhwemhal2::HwEmShim::xclReadModifyWrite(uint64_t, const void*, size_t)’:
/home/xilinx/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx:1574:67: error: requested alignment 128 is larger than 16 [-Werror=attributes]
     alignas(DDR_BUFFER_ALIGNMENT) char buffer[DDR_BUFFER_ALIGNMENT];
                                                                   ^

Please refer to attached as full installation log. Do you have any suggestion?
Thanks

Mike

sonals, 2018-12-20:
Fixed by
https://github.com/Xilinx/XRT/pull/815

cjvolzka, 2019-01-03:
After fix #815 I'm still getting a similar error trying to build on ppc64le as well.
```
[ 62%] Building CXX object runtime_src/driver/xclng/tools/xbflash/CMakeFiles/xbflash.dir/flasher.cpp.o
/root/cjvolzka/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx: In member function ‘size_t xclhwemhal2::HwEmShim::xclReadSkipCopy(uint64_t, void*, size_t)’:
/root/cjvolzka/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx:1542:67: error: requested alignment 128 is larger than 16 [-Werror=attributes]
     alignas(DDR_BUFFER_ALIGNMENT) char buffer[DDR_BUFFER_ALIGNMENT];
                                                                   ^
[ 62%] Building CXX object runtime_src/driver/xclng/tools/xbflash/CMakeFiles/xbflash.dir/prom.cpp.o
/root/cjvolzka/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx: In member function ‘size_t xclhwemhal2::HwEmShim::xclReadModifyWrite(uint64_t, const void*, size_t)’:
/root/cjvolzka/XRT/src/runtime_src/driver/hw_em/generic_pcie_hal2/shim.cxx:1584:67: error: requested alignment 128 is larger than 16 [-Werror=attributes]
     alignas(DDR_BUFFER_ALIGNMENT) char buffer[DDR_BUFFER_ALIGNMENT];

....

[ 99%] Built target xrt_swemu
cc1plus: all warnings being treated as errors
runtime_src/driver/hw_em/CMakeFiles/hw_emu_objects.dir/build.make:182: recipe for target 'runtime_src/driver/hw_em/CMakeFiles/hw_emu_objects.dir/generic_pcie_hal2/shim.cxx.o' failed
make[2]: *** [runtime_src/driver/hw_em/CMakeFiles/hw_emu_objects.dir/generic_pcie_hal2/shim.cxx.o] Error 1
CMakeFiles/Makefile2:1633: recipe for target 'runtime_src/driver/hw_em/CMakeFiles/hw_emu_objects.dir/all' failed
make[1]: *** [runtime_src/driver/hw_em/CMakeFiles/hw_emu_objects.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 99%] Built target xclbinutil
[ 99%] Linking CXX shared library libxdp.so
[ 99%] Built target xclbincat
[ 99%] Built target xdp
[100%] Linking CXX executable xbutil
[100%] Built target xbutil
[100%] Built target xocl
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

The #815 fix was to disable emulation library builds on ppc64le. However commit https://github.com/Xilinx/XRT/commit/0850c9f6ecda930144462282cbeae146614f99db#diff-cd3212400dccc73763a8a00cfe42cd88 appears to have undone that. I see in the build output 
```
-- Host system processor is ppc64le
-- Target system processor is ppc64le
```
Based on the commit that means XRT_NATIVE_BUILD should be "yes" causing the emulation libraries to be built again on ppc64le. I'm unsure if either those are now supposed to work on ppc64le or if building them again was an unintended side effect.

Extra note:
I tried setting src/runtime_src/driver/CMakeLists.txt line 2 to `if ((${XRT_NATIVE_BUILD} STREQUAL "yes") AND (${CMAKE_SYSTEM_PROCESSOR} STREQUAL x86_64))` to replicate fix #815 and re-disable the emulation libraries. However that still failed with
```
[ 32%] Building CXX object runtime_src/xdp/CMakeFiles/xdpobj.dir/debug/rt_debug.cpp.o
/root/cjvolzka/XRT/src/runtime_src/driver/zynq/user/shim.cpp:78:16: error: trigraph ??) ignored, use -trigraphs to enable [-Werror=trigraphs]
     status = "(??)";
```

Full logs from both build attempts:
[unmodified build fail.log](https://github.com/Xilinx/XRT/files/2724476/unmodified.build.fail.log)
[modified verbose build.log](https://github.com/Xilinx/XRT/files/2724477/modified.verbose.build.log)

sonals, 2019-01-17:
Can you try again? I compiled 2018.3 and master today on PPC64LE without any issues.

sonals, 2019-01-18:
Thanks closing the issue.

closed 2019-01-18 by sonals