Xilinx/XRT #813: CMAKE_INSTALL_PREFIX not respected

pwaller, 2018-12-19:
It would be helpful to people trying to package XRT if `CMAKE_INSTALL_PREFIX` were respected.

Right now it is not:

```
cmake \
        -DCMAKE_INSTALL_PREFIX=/usr/local \
        -DCMAKE_BUILD_TYPE=Release
        /path/to/src
```

Results in files in `/usr/src`, and files in /opt/xilinx, outside the given prefix:

<details>

<summary>List of files installed</summary>

```
Install the project...
-- Install configuration: "Release"
-- Installing: /opt/xilinx/xrt/include/CL/cl_ext_xilinx.h
-- Installing: /opt/xilinx/xrt/include/CL/cl_ext.h
-- Installing: /opt/xilinx/xrt/license/LICENSE
-- Installing: /opt/xilinx/xrt/include/version.h
-- Installing: /usr/src/xrt-2.1.0/driver/include/version.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-core.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-cw.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-utils.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-ioctl.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-sysfs.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/mgmt-core.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/xclmgmt.dracut.conf
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/10-xclmgmt.rules
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/mgmtpf/Makefile
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/devices.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/xocl_drv.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/xocl_subdev.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/xocl_ctx.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/xocl_thread.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/xocl_test.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xdma.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/qdma.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/common.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_bo.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_bo.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_drm.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_drm.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_ioctl.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_sysfs.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl_drv.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/xocl.dracut.conf
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/10-xocl.rules
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/userpf/Makefile
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libxdma.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libxdma.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/cdev_sgdma.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libxdma_api.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/Makefile.in
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/libqdma_export.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/libqdma_export.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_descq.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_descq.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_device.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_device.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_intr.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_intr.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_context.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_context.h
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_mbox.c
-- Installing: /usr/src/xrt-2.1.0/driver/xclng/drm/xocl/lib/libqdma/qdma_mbox.h
[... 122 lines cut ...]
-- Installing: /opt/xilinx/xrt/include/app/xmafilter.h
-- Installing: /opt/xilinx/xrt/include/app/xmascaler.h
-- Installing: /opt/xilinx/xrt/include/app/xmakernel.h
-- Installing: /opt/xilinx/xrt/include/app/xmabuffers.h
-- Installing: /opt/xilinx/xrt/include/app/xmalogger.h
-- Installing: /opt/xilinx/xrt/include/app/xmadecoder.h
-- Installing: /opt/xilinx/xrt/lib/libxmaplugin.so.2.1.0
-- Installing: /opt/xilinx/xrt/lib/libxmaplugin.so.2
-- Installing: /opt/xilinx/xrt/lib/libxmaplugin.so
-- Installing: /opt/xilinx/xrt/lib/libxmaapi.so.2.1.0
-- Installing: /opt/xilinx/xrt/lib/libxmaapi.so.2
-- Installing: /opt/xilinx/xrt/lib/libxmaapi.so
```

</details>

---

I do understand that there may be constraints on where files must live (e.g./etc/OpenCL/vendors/xilinx.icd) in order for the runtime to actually execute. I'm happy to sort those constraints out later in the packaging process. (But ideally given that this file just references `libxilinxopencl.so`, it seems to me that this should be arranged to point to the correct place).

In my ideal world, if I specify `CMAKE_INSTALL_PREFIX`, then libraries and headers should all appear at `${CMAKE_INSTALL_PREFIX}/{bin,lib,lib/pkgconfig,include}`.

I have tried with `make DESTDIR=/usr/local install`, and this results in files 

See also #86.

keryell, 2019-06-04:
The problem I see is that the generated `.deb` contains a lot of stuff, which is usually uncommon.
Probably a more common use case would be to have a top-level `xrt.deb` mainly empty, just pulling a lot of sub `.deb` though its dependencies, such as `xrt-kernel-source`, `xrt-bin`, `xrt-lib`, `xrt-doc`, `xrt-dev`, `xrt-icd`, etc. But that would require a lot of devop re-hauling...
@pwaller I do not know then if it would make sense to have only 1 top-level `CMAKE_INSTALL_PREFIX` or one per sub-project...

uday610, 2021-07-30:
obsolated

closed 2021-07-30 by uday610