Xilinx/XRT #836: build issue with cmake 3.13.2

tianhaoz95-xilinx, 2019-01-02:
OS: Ubuntu 16.04
System Spec:
```bash
uname -a
Linux 4.4.0-138-generic #164-Ubuntu SMP Tue Oct 2 17:16:02 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
cmake version:
```bash
cmake --version
cmake version 3.13.2
CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
Error message:
```bash
/bin/sh: ./kernel-doc: No such file or directory
runtime_src/doc/CMakeFiles/xrt_docs.dir/build.make:79: recipe for target 'runtime_src/doc/core/ert.rst' failed
make[2]: *** [runtime_src/doc/core/ert.rst] Error 127
make[2]: *** Deleting file 'runtime_src/doc/core/ert.rst'
make[2]: *** Waiting for unfinished jobs....
[  9%] Generating core/mgmt-ioctl.rst
/bin/sh: ./kernel-doc: No such file or directory
runtime_src/doc/CMakeFiles/xrt_docs.dir/build.make:67: recipe for target 'runtime_src/doc/core/mgmt-ioctl.rst' failed
make[2]: *** [runtime_src/doc/core/mgmt-ioctl.rst] Error 127
make[2]: *** Deleting file 'runtime_src/doc/core/mgmt-ioctl.rst'
Scanning dependencies of target xclbinutil
CMakeFiles/Makefile2:537: recipe for target 'runtime_src/doc/CMakeFiles/xrt_docs.dir/all' failed
make[1]: *** [runtime_src/doc/CMakeFiles/xrt_docs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

mamin506, 2019-02-13:
@tianhaoz95-xilinx Do you still see this issue in the latest code?
Could you check below items:
1. internet access on that build machine
2. "./kernel-doc downloading" message in the cmake log
3. build/Release/kernel-doc exist or not

tianhaoz95-xilinx, 2019-02-19:
I haven't been using that machine for a while. Will give it a try.

tianhaoz95, 2019-03-07:
@mamin506 Yes I still see this issue in the latest code.

error message:

```
/bin/sh: ./kernel-doc: No such file or directory
runtime_src/doc/CMakeFiles/xrt_docs.dir/build.make:89: recipe for target 'runtime_src/doc/core/xmaplugin.rst' failed
make[2]: *** [runtime_src/doc/core/xmaplugin.rst] Error 127
make[2]: *** Deleting file 'runtime_src/doc/core/xmaplugin.rst'
CMakeFiles/Makefile2:635: recipe for target 'runtime_src/doc/CMakeFiles/xrt_docs.dir/all' failed
make[1]: *** [runtime_src/doc/CMakeFiles/xrt_docs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

1. internet access on that build machine

I can `git pull` and `git clone`, but the mirror for kernel-doc is hard to say.

2. "./kernel-doc downloading" message in the CMake log

Yes, I see the output message without tailing error.

```
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- ./kernel-doc downloading
-- Boost version: 1.58.0
```

3. build/Release/kernel-doc exist or not

Yes, it exists.

```
ls ./build/Release/
BMC-MSP432.found.txt  CMakeFiles	   cmc.bin.found.txt	  CPackConfig.cmake	   dkms-awsmgmt  gen	     libxmaapi.pc     Makefile	postinst-aws  prerm-aws  runtime_src  xma	  xrt.pc
CMakeCache.txt	      cmake_install.cmake  compile_commands.json  CPackSourceConfig.cmake  dkms.conf	 kernel-doc  libxmaplugin.pc  postinst	prerm	      python	 xilinx.icd   xrt-aws.pc
```

uday610, 2019-03-26:
@tianhaoz95 , is this issue exist now after #1055 is closed?

tianhaoz95-xilinx, 2019-04-04:
@uday610 yes this can be closed

closed 2019-04-04 by tianhaoz95-xilinx