Xilinx/XRT #1136: Which version of Linux kernel XRT device driver can be compiled for?

keryell, 2019-03-26:
By looking at https://github.com/Xilinx/XRT/blob/master/src/runtime_src/doc/toc/system_requirements.rst I understand that there is a list of Linux distributions XRT has been tested for.
But is there a list of Linux kernel versions the device driver is know to compile on?
Such as what is the most recent version of Linux (5.0?) and the oldest?
It would be nice to add a new section for the Linux kernels in this document, along the distributions.
This is important if we want a wider use of XRT.

keryell, 2019-04-15:
It would be nice to add something like Travis-CI to check that it compiles with a few recent kernel versions and a few GCC versions... I am not even thinking about checking it does really work on a real system and real device. Just checking that it compiles without scary warning is a first big step. :-)

Right now the branch `2018-3` is XRT 2.1.0 compiles for me with Linux 4.18.0-17-generic (x86_64) and GCC (Ubuntu 8.2.0-7ubuntu1) version 8.2.0 (x86_64-linux-gnu) from Ubuntu 18.10, but the `master`branch which is XRT 2.2.0 does not compile in the same environment...

I have not tried yet with the brand new Ubuntu 19.04 or Debian/sid.

keryell, 2019-05-22:
To move on this, someone inside Xilinx can try to compile on `xsjsycl40.xilinx.com` if you need a machine with Ubuntu 18.10 &  Linux 4.18. I can add you in the sudoers too.

sonals, 2021-11-05:
Closing as obsolete.

closed 2021-11-05 by sonals