Xilinx/XRT #1263: xbutil crash on flashing

hackwa, 2019-04-19:
xbutil flash -a xilinx_u200_xdma_201830_1
XBFLASH -- Xilinx Card Flash Utility
Probing card[0]: Shell on FPGA needs updating
Shell on below card(s) will be updated:
Card [0]
Are you sure you wish to proceed? [y/n]
y
Updating shell on card[0]
INFO: ***Found 701 ELA Records
Idcode byte[0] ff
Idcode byte[1] 20
Idcode byte[2] bb
Idcode byte[3] 21
Idcode byte[4] 10
Enabled bitstream guard. Bitstream will not be loaded until flashing is finished.
Erasing flash........................z...........
Programming flashxbutil: /root/hackwad/XRT/src/runtime_src/driver/xclng/tools/xbmgmt/xspi.cpp:1359: int XSPI_Flasher::programXSpi(std::istream&, const XSPI_Flasher::ELARecord&): Assertion `(address + da
taLen - (record.mStartAddress & 0xFFFF)) == static_cast<unsigned int>((pageIndex +1)*128)' failed.
Aborted (core dumped)

maxzhen, 2019-04-19:
@cellery , do you know anything about this assertion?

cellery, 2019-04-23:
This assertion fails when you try to flash with debug XRT build and not release. No one ever updated the assert statement to fix the check.

keryell, 2019-06-04:
@sonals who is the right person to move on this?
I just spent a couple hours on this again and that is a shame. :-(
Is there any Continuous Integration made on this?
Who is in charge of this CI inside Xilinx?

maxzhen, 2019-06-04:
Did you try latest master branch? Should've been fixed there.

keryell, 2019-06-04:
The problem is that I do not know if I can use the `master` branch or not. Is it supposed to work with the SDx 2019.1 public version? I am going to create an issue on it.
For now I was trying only officially available stuff...

maxzhen, 2019-06-04:
the master should work. if not, it's a bug of highest priority.

hackwa, 2019-06-07:
Awesome! thanks for confirming @keryell. I do not know what the specific commit is for cherry picking. Maybe @maxzhen  or @cellery could help?

maxzhen, 2019-06-07:
The change is just to remove the assertion. Can't do cherry-pick since it's part of a big change which can't be backported to 2019.1. For 2019.1, please use release version of xbutil flash.

closed 2019-06-07 by hackwa