Xilinx/XRT #1287: Compilation hang on Ubuntu 16

tianhaoz95, 2019-04-23:
OS: Ubuntu 16
Did I run `xrtdeps.sh`: yes

Error message:
```bash
-- XRT EA eula files  /proj/isi_group/isim/tianhaoz/Github/XRT_dev/rtl-no-profiling/XRT/src/../LICENSE
-- Platform/Linux (Ubuntu) (Kernel 4.4.0-138-generic)
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_MAIN_LIBRARY) 
-- GTest was not found, skipping generation of test executables
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
-- Rodin devkits found, preparing ERT build
CMake Warning at runtime_src/ert/cmc/CMakeLists.txt:24 (MESSAGE):
  Skipped cmc.bin packaging with error code 28
```

keryell, 2019-06-04:
Is this problem solved for you?
Otherwise, Ubuntu 16.04 is 3 year old anyway, what about modernizing your environment?

tianhaoz95, 2019-06-04:
I tried on another Ubuntu 16.04, the same issue did not show up. I think it is a problem with the corporate proxy. Closing this issue.

closed 2019-06-04 by tianhaoz95