Xilinx/XRT #1384: Device driver fails to compile with Linux 5.0

keryell, 2019-05-08:
I have tried with branches `2018.3` and `2019.1` on Ubuntu 19.04 and it fails with messages like:
```
DKMS make.log for xrt-2.1.0 for kernel 5.0.0-13-generic (x86_64)
Wed 08 May 2019 02:13:46 PM PDT
cd userpf; make all
make[1]: Entering directory '/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf'
echo /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf
/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf
make -C /lib/modules/5.0.0-13-generic/build M=/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf modules
make[2]: Entering directory '/usr/src/linux-headers-5.0.0-13-generic'
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../xocl_subdev.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../xocl_ctx.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../xocl_thread.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mm_xdma.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/feature_rom.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mm_qdma.o
  CC [M]  /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mb_scheduler.o
/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mb_scheduler.c: In function ‘cmd_release_gem_object_reference’:
/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mb_scheduler.c:409:3: error: implicit declaration of function ‘drm_gem_object_unreference_unlocked’; did you mean ‘drm_gem_object_put_unlocked’? [-Werror=implicit-function-declaration]
   drm_gem_object_unreference_unlocked(&xcmd->bo->base);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   drm_gem_object_put_unlocked
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:286: /var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf/../subdev/mb_scheduler.o] Error 1
make[2]: *** [Makefile:1584: _module_/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.0.0-13-generic'
make[1]: *** [Makefile:57: all] Error 2
make[1]: Leaving directory '/var/lib/dkms/xrt/2.1.0/build/driver/xclng/drm/xocl/userpf'
make: *** [Makefile:2: all] Error 2
```
It looks like `drm_gem_object_unreference_unlocked` has been renamed to `drm_gem_object_put_unlocked` in Linux 5.0.
 
Interestingly `drm_gem_object_put_unlocked` is used already in `src/runtime_src/driver/zynq/drm/zocl/zocl_bo.c`#641 starting with branch 2019.1:
```
	drm_gem_object_put_unlocked(&cma_obj->base);
```
I am puzzled here...
How can it work with 2 API at the same time that cannot coexist?

keryell, 2019-05-22:
I am setting up a machine with a Ubuntu 19.04 and a U200 so someone from the team can play with...

keryell, 2019-05-22:
Actually compiling for Linux 5.0 does not need a specific machine up to the final test I can do myself...
But if someone can fix the compilation issue in the meantime, without testing, it is a good start... :-)

keryell, 2019-06-04:
@sonals who is the right person to answer this?

j-stephan, 2019-06-06:
@keryell: See #1530, it seems to work on my machine.

j-stephan, 2019-06-11:
It should be fixed now, give it a try :)

uday610, 2019-07-09:
fixed

closed 2019-07-09 by uday610