Xilinx/XRT #1492: Utils don't see BCU 1525

trappist, 2019-05-30:
I originally opened #818 to this effect but it was closed after I neglected it. Still have the same problem. Re: the comments on that issue, `lspci` shows the device as `1e24:1525`, not `10ee:*` which I assume is the source of the problem. The `xocl` and `xclmgmt` are both loaded according to `lsmod`. This is the relevant output of lspci, if it helps:
```
06:00.0 Co-processor: Device 1e24:1525 (rev ff)
07:00.0 Co-processor: Device 1e24:1525 (rev ff)
08:00.0 Co-processor: Device 1e24:1525 (rev ff)
0a:00.0 Co-processor: Device 1e24:1525 (rev ff)
0e:00.0 Co-processor: Device 1e24:1525 (rev ff)
0f:00.0 Co-processor: Device 1e24:1525 (rev ff)
11:00.0 Co-processor: Device 1e24:1525 (rev ff)
12:00.0 Co-processor: Device 1e24:1525 (rev ff)
13:00.0 Co-processor: Device 1e24:1525 (rev ff)
15:00.0 Co-processor: Device 1e24:1525 (rev ff)
16:00.0 Co-processor: Device 1e24:1525 (rev ff)
17:00.0 Co-processor: Device 1e24:1525 (rev ff)
18:00.0 Co-processor: Device 1e24:1525 (rev ff)
19:00.0 Co-processor: Device 1e24:1525 (rev ff)
```

uday610, 2019-07-08:
Hello @trappist , the BCU 1525 is not a Xilinx device. As per this page : https://pci-ids.ucw.cz/read/PC/1e24 this device is from  Squirrels Research Labs. So Xilinx cannot support this and you may need to contact to the correct vendor.

trappist, 2019-07-08:
OK thanks @uday610

closed 2019-07-08 by trappist