Xilinx/XRT #1497: cannot flush outstanding command during tests/xrt/102_multiproc_verify test

chaosYoun, 2019-05-30:
[  240.352866] xocl 0000:59:00.1: exec_stop: exec_stop(000000007844613e)
[  240.352872] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.457552] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.565562] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.673586] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.781559] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.889563] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  240.997556] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.105571] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.213560] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.321558] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.429555] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.537556] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.645552] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.753553] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.861552] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  241.969550] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  242.077550] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  242.185549] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  242.293547] xocl 0000:59:00.1: exec_stop: Waiting for 1 outstanding commands to finish
[  243.413543] xocl 0000:59:00.1: exec_stop: unexpected outstanding commands 1 after flush

Somtimes last cmd_free() can call exec_stop() through cmd_release_gem_object_reference().
Should decrease outstanding_exec count first to avoid unexpected outstanding commands.

@@ -563,14 +563,14 @@ cmd_get(struct xocl_scheduler *xs, struct exec_core *exec, struct client_ctx *cl
 static void
 cmd_free(struct xocl_cmd *xcmd)
 {
-       cmd_release_gem_object_reference(xcmd);
-
        mutex_lock(&free_cmds_mutex);
        list_move_tail(&xcmd->cq_list, &free_cmds);
        mutex_unlock(&free_cmds_mutex);

        atomic_dec(&xcmd->xdev->outstanding_execs);
        atomic_dec(&xcmd->client->outstanding_execs);
+
+       cmd_release_gem_object_reference(xcmd);
        SCHED_DEBUGF("xcmd(%lu) [-> free]\n", xcmd->uid);
 }

chaosYoun, 2019-05-31:
Below code also can cause similar problem.
```diff
@@ -2410,8 +2410,8 @@ static void
 scheduler_error_to_free(struct xocl_scheduler *xs, struct xocl_cmd *xcmd)
 {
        SCHED_DEBUGF("-> %s(%lu)\n", __func__, xcmd->uid);
-       exec_notify_host(cmd_exec(xcmd));
        scheduler_complete_to_free(xs, xcmd);
+       exec_notify_host(cmd_exec(xcmd));
        SCHED_DEBUGF("<- %s\n", __func__);
 }

stsoe, 2019-07-17:
I second the change in scheduler_error_to_free.  But I am not comfortable with cmd_free change.   Once the xocl_cmd is freed it will be recycled through cmd_get where the ->bo field is reset.   If cmd_release_gem_object is deferred as suggested, the reset could come before the bo was released to drm.

Of course the free_cmd_mutex could be held longer, but I am curious about the scenario you describe.  How will cmd_release_gem_object indirectly call exec_stop?

chaosYoun, 2019-07-17:
> I am curious about the scenario you describe. How will cmd_release_gem_object indirectly call exec_stop?
I think my description can be wrong.
I thought that exec_stop() was called somehow by cmd_release_gem_object_reference() at that time but exec_stop() may be called before cmd_free() is called through exec_mark_cmd_complete().
For this problem to happen,  cmd_release_gem_object_reference() should be blocked by some other reason like mutex.

PR #1705 (2019-07-23): Multi process fixes for aborted commands
Fixed a few problems related to client release of context after
command failure. Client must be notified *after* scheduler has updated
its bookkeeping related to outstanding commands.

Notification can result in bitstream unlock.  Bitstream unlock
acquires device mutex lock, calls exec_stop() which waits for command
queue to drain.  If order of notification and bookkeeeping is reversed
then its pssible bookkeeping is behind at the time exec_stop is called.

Fixes: #1497
commit 697c73ce (2019-07-23): Multi process fixes for aborted commands (#1705)

Fixed a few problems related to client release of context after
command failure. Client must be notified *after* scheduler has updated
its bookkeeping related to outstanding commands.

Notification can result in bitstream unlock.  Bitstream unlock
acquires device mutex lock, calls exec_stop() which waits for command
queue to drain.  If order of notification and bookkeeeping is reversed
then its pssible bookkeeping is behind at the time exec_stop is called.

Fixes: #1497
closed 2019-07-23 by stsoe