Xilinx/XRT #1498: Shouldn't abort the command which was already completed

chaosYoun, 2019-05-30:
In ERT mode, scheduler_running_to_complete() can complete multiple commands.
Theses completed commands can be remain as completed even after client->abort is set as true.
Should not change the state of these commands to ERT_CMD_STATE_ABORT or application can get false error return.
 
diff --git a/src/runtime_src/core/pcie/driver/linux/xocl/subdev/mb_scheduler.c b/src/runtime_src/core/pcie/driver/linux/xocl/subdev/mb_scheduler.c
index 143c66a..bb888f6 100644
--- a/src/runtime_src/core/pcie/driver/linux/xocl/subdev/mb_scheduler.c
+++ b/src/runtime_src/core/pcie/driver/linux/xocl/subdev/mb_scheduler.c
@@ -411,7 +411,7 @@ cmd_set_state(struct xocl_cmd *xcmd, enum ert_cmd_state state)
 static enum ert_cmd_state
 cmd_update_state(struct xocl_cmd *xcmd)
 {
-       if (xcmd->state != ERT_CMD_STATE_RUNNING && xcmd->client->abort) {
+       if (xcmd->state != ERT_CMD_STATE_RUNNING && xcmd->state != ERT_CMD_STATE_COMPLETED && xcmd->client->abort) {
                userpf_info(xcmd->xdev, "aborting stale client cmd(%lu)", xcmd->uid);
                cmd_set_state(xcmd, ERT_CMD_STATE_ABORT);
        }

stsoe, 2019-07-17:
Hi.  Thanks for this suggestion.   client->abort is set to true when client / application exists.  How can application get false return code?

chaosYoun, 2019-07-17:
client->abort is set as true in destrory_client().
After setting client->abort as true, destroy_client() is waiting for all the outstanding commands to finish.
I don't remember exact call stack of the 2 threads but based on the code, there is a possibility that while the thread calling destroy_client() waits for the outstanding commands, the scheduler thread changes the command state and notifies the application.

chaosYoun, 2019-07-17:
For your information, I found this error during tests/xrt/102_multiproc_verify test.

stsoe, 2019-07-18:
> there is a possibility that while the thread calling destroy_client() waits for the outstanding commands, the scheduler thread changes the command state and notifies the application.

But my point is that the application must already be in exit handlers when destroy_client() is called, thus the application can no longer get notifications.

chaosYoun, 2019-07-18:
Today morning I tried to reproduce the symptom I described without the patch but failed.
Your point seems reasonable and I don't remember the details of this issue now.
I'll close this and reopen if I encounter the same issue.

closed 2019-07-18 by chaosYoun