Xilinx/XRT #1606: unable to perform ./build.sh

durgabhavaniv, 2019-07-02:
cd XRT/build
./build.sh

Above step throwing the following error as below:
/usr/bin/cmake3: /opt/Xilinx/SDx/2018.2/lib/lnx64.o/libxml2.so.2: no version information available (required by /lib64/libarchive.so.13)

keryell, 2019-07-02:
You should have a clean environment with the required packages installed/compiled.
So, *do not* source some configuration files from Xilinx `/opt/Xilinx/...` or `/proj/xbuilds/...` otherwise you can get some random pieces of bits...

durgabhavaniv, 2019-07-03:
With out sourcing I got same issue as below:

[ 45%] Linking CXX executable xclbintest
/usr/bin/cmake3: /opt/Xilinx/SDx/2018.2/lib/lnx64.o/libxml2.so.2: no version information available (required by /lib64/libarchive.so.13)
[ 45%] Built target xclbinutil
/usr/bin/cmake3: /opt/Xilinx/SDx/2018.2/lib/lnx64.o/libxml2.so.2: no version information available (required by /lib64/libarchive.so.13)
[ 45%] Built target xclbintest
make: *** [all] Error 2

keryell, 2019-07-04:
Curious...
Where does this `/opt/Xilinx/SDx/2018.2` come from then?
What is the output of
```
printenv | grep /opt/Xilinx/SDx
```
for example?

What is your OS? What is this `cmake3`?

uday610, 2019-07-22:
There is no response from the reporter. So closing this

closed 2019-07-22 by uday610