Xilinx/XRT #1717: Name clashes with Linux kernel >= 5.0.14

j-stephan, 2019-07-25:
Linux 5.0.14 introduced `stream_open` in [`include/linux/fs.h`](https://elixir.bootlin.com/linux/v5.0.14/source/include/linux/fs.h#L3080). This causes a name clash while building the kernel module since `stream_open` is already defined there:

```
/var/lib/dkms/xrt/2.3.0/build/driver/xocl/userpf/../subdev/qdma.c:1712:12: error: static declaration of ‘stream_open’ follows non-static declaration
 static int stream_open(struct inode *inode, struct file *file)
            ^~~~~~~~~~~
In file included from ./include/linux/huge_mm.h:8,
                 from ./include/linux/mm.h:526,
                 from ./include/linux/scatterlist.h:8,
                 from ./include/linux/dma-mapping.h:11,
                 from ./include/drm/drmP.h:37,
                 from /var/lib/dkms/xrt/2.3.0/build/driver/xocl/userpf/../subdev/qdma.c:24:
./include/linux/fs.h:3095:12: note: previous declaration of ‘stream_open’ was here
 extern int stream_open(struct inode * inode, struct file * filp);
            ^~~~~~~~~~~
```

I'd submit a pull request and rename this myself but I don't know about Xilinx' internal coding standards / naming guidelines.

maxzhen, 2019-07-25:
maybe prefix it with "qdma" or something as other functions in the same file is using?

j-stephan, 2019-07-25:
Okay, I simply renamed all `stream_foo` structs / functions to `qdma_stream_foo`. See #1720.

keryell, 2019-07-28:
Thanks!
@sonals It would be nice to have a buildbot at Xilinx testing the compilation with the recent kernels if we are serious about open-source...
I guess this can be closed since https://github.com/Xilinx/XRT/pull/1720 was merged.

j-stephan, 2019-07-31:
Yup, closing this, compilation works fine (for now...)

closed 2019-07-31 by j-stephan