Xilinx/XRT #1775: XRT 2.2.2173 does not support RHEL 7.7

dsorber, 2019-08-06:
With the latest version of XRT 2.2.2173 the xocl driver does not compile on RHEL 7.7:

```
DKMS make.log for xrt-2.2.2173 for kernel 3.10.0-1062.el7.x86_64 (x86_64)
Tue Aug  6 16:29:16 EDT 2019
cd userpf; make all
make[1]: Entering directory `/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf'
echo /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf
make -C /lib/modules/3.10.0-1062.el7.x86_64/build M=/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf modules
make[2]: Entering directory `/usr/src/kernels/3.10.0-1062.el7.x86_64'
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_subdev.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_ctx.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_thread.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/xdma.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.o
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c: In function ‘queue_req_complete’:
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c:931:3: error: implicit declaration of function ‘drm_gem_object_unreference_unlocked’ [-Werror=implicit-function-declaration]
   drm_gem_object_unreference_unlocked(&cb->xobj->base);
   ^
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c: In function ‘stream_post_bo’:
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c:965:2: error: implicit declaration of function ‘drm_gem_object_reference’ [-Werror=implicit-function-declaration]
  drm_gem_object_reference(gem_obj);
  ^
cc1: some warnings being treated as errors
make[3]: *** [/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.o] Error 1
make[2]: *** [_module_/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf] Error 2
make[2]: Leaving directory `/usr/src/kernels/3.10.0-1062.el7.x86_64'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf'
make: *** [all] Error 2
~                         
```

dsorber, 2019-08-16:
Is there any update on this?  I still can't use my U200 card until the driver is updated.

houlz0507, 2019-08-16:
kernel 5.0 compiling fix is back ported to 19.1 branch. Could you try the latest 19.1 branch on git hub?
It should work but we do not have 3.10.0-1062 to verify it. Please let us know if you still can not compile.

dsorber, 2019-08-16:
I just built the 19.1 branch from source and installed the resulting RPM on 3.10.0-1062.  The driver modules compiled and installed successfully. 

Do you know when an official updated RPM will be released?

houlz0507, 2019-08-16:
I do not know the next release time. I am copying @uday610 for your question

dsorber, 2019-08-19:
Thanks for your attention to this issue. RHEL 7.7 (with the 3.10.0-1062 kernel) was released on 8/6, and CentOS releases usually lag RHEL releases by about 30 days.  I strongly suggest that a new release containing the driver updates be done before CentOS 7.7 is released. 

I can use the custom built XRT RPM for my own development purposes but I need the updated official RPM for any customer deployments.

dsorber, 2019-08-22:
@houlz0507 I attempted to do some more testing with the XRT RPM I built from the 19.1 branch.  I can't seem to load firmware (.xclbin files) with the new version of XRT installed.

This is what I tried, the command hangs:

```
$ xbutil program -p firmware.xclbin
INFO: Found total 1 card(s), 1 are usable
```
This is the contents of the kernel log:
```
[Thu Aug 22 11:07:15 2019] [drm] creating scheduler client for pid(15644), ret: 0
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:15 2019] icap.u icap.u.24065: icap_read_from_peer: reading from peer
[Thu Aug 22 11:07:15 2019] mailbox.u mailbox.u.24065: mailbox_request: sending request: 10 via HW
[Thu Aug 22 11:07:15 2019] mailbox.m mailbox.m.24064: process_request: received request from peer: 10, passed on
[Thu Aug 22 11:07:15 2019] mailbox.m mailbox.m.24064: mailbox_post_response: posting response for: 10 via HW
[Thu Aug 22 11:07:15 2019] [drm] client exits pid(15644)
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_close: NOTIFY ffff9226f9838810
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: char_open: opened file ffff9256deea1b00 by pid: 15644
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: char_close: Closing file ffff9256deea1b00 by pid: 15644
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: NOTIFY ffff9226e3eda810
[Thu Aug 22 11:07:15 2019] [drm] creating scheduler client for pid(15644), ret: 0
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_mmap: io_remap_pfn_range ret code: 0
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: char_open: opened file ffff9256deea2400 by pid: 15644
[Thu Aug 22 11:07:15 2019] xmc.u xmc.u.24065: xmc_read_from_peer: reading from peer
[Thu Aug 22 11:07:15 2019] mailbox.u mailbox.u.24065: mailbox_request: sending request: 10 via HW
[Thu Aug 22 11:07:15 2019] mailbox.m mailbox.m.24064: process_request: received request from peer: 10, passed on
[Thu Aug 22 11:07:15 2019] mailbox.m mailbox.m.24064: mailbox_post_response: posting response for: 10 via HW
[Thu Aug 22 11:07:15 2019] [drm] client exits pid(15644)
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:15 2019] xocl 0000:5e:00.1: xocl_drvinst_close: NOTIFY ffff9226f9838810
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: char_close: Closing file ffff9256deea2400 by pid: 15644
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:15 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: NOTIFY ffff9226e3eda810
[Thu Aug 22 11:07:44 2019] [drm] creating scheduler client for pid(15648), ret: 0
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:44 2019] icap.u icap.u.24065: icap_read_from_peer: reading from peer
[Thu Aug 22 11:07:44 2019] mailbox.u mailbox.u.24065: mailbox_request: sending request: 10 via HW
[Thu Aug 22 11:07:44 2019] mailbox.m mailbox.m.24064: process_request: received request from peer: 10, passed on
[Thu Aug 22 11:07:44 2019] mailbox.m mailbox.m.24064: mailbox_post_response: posting response for: 10 via HW
[Thu Aug 22 11:07:44 2019] [drm] client exits pid(15648)
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_drvinst_close: NOTIFY ffff9226f9838810
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: char_open: opened file ffff9226e6cee300 by pid: 15648
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: char_close: Closing file ffff9226e6cee300 by pid: 15648
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: CLOSE 2
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_close: NOTIFY ffff9226e3eda810
[Thu Aug 22 11:07:44 2019] [drm] creating scheduler client for pid(15648), ret: 0
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_mmap: io_remap_pfn_range ret code: 0
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: xocl_drvinst_open: OPEN 1
[Thu Aug 22 11:07:44 2019] xclmgmt 0000:5e:00.0: char_open: opened file ffff9226e6cefd00 by pid: 15648
[Thu Aug 22 11:07:44 2019] xmc.u xmc.u.24065: xmc_read_from_peer: reading from peer
[Thu Aug 22 11:07:44 2019] mailbox.u mailbox.u.24065: mailbox_request: sending request: 10 via HW
[Thu Aug 22 11:07:44 2019] mailbox.m mailbox.m.24064: process_request: received request from peer: 10, passed on
[Thu Aug 22 11:07:44 2019] mailbox.m mailbox.m.24064: mailbox_post_response: posting response for: 10 via HW
[Thu Aug 22 11:07:44 2019] xocl 0000:5e:00.1: xocl_read_axlf_helper: READ_AXLF IOCTL
[Thu Aug 22 11:07:44 2019] rom.u rom.u.24065: verify_timestamp: DSA timestamp: 0x5d3e3fa4
[Thu Aug 22 11:07:44 2019] rom.u rom.u.24065: verify_timestamp: Verify timestamp: 0x5d3e3fa4
[Thu Aug 22 11:07:44 2019] XOCL: VBNV and TimeStamps matched
[Thu Aug 22 11:07:44 2019] XOCL: Marker 5
[Thu Aug 22 11:07:44 2019] [drm] Finding MEM_TOPOLOGY section header
[Thu Aug 22 11:07:44 2019] [drm] Section MEM_TOPOLOGY details:
[... cut ...]
[Thu Aug 22 11:07:47 2019] xocl 0000:5e:00.1: cu_reset: configured cu(3) base@0x1e10000 poll@0x0 control(0)
[Thu Aug 22 11:07:47 2019] xocl 0000:5e:00.1: cu_reset: configured cu(4) base@0x250000 poll@0x0 control(0)
[Thu Aug 22 11:07:47 2019] xocl 0000:5e:00.1: exec_cfg_cmd: configuring embedded scheduler mode
[Thu Aug 22 11:07:47 2019] xocl 0000:5e:00.1: exec_cfg_cmd: scheduler config ert(1), dataflow(0), slots(16), cudma(1), cuisr(1), cdma(1), cus(5)
```
Do you have any idea what's wrong, or any thoughts on what I can do to debug? The same test with XRT version 2.1.1794, works just fine.

houlz0507, 2019-08-22:
Are you saying that "xbutil program -p firmware.xclbin" hangs? If yes, maybe @stsoe has some insight?

dsorber, 2019-08-22:
> Are you saying that "xbutil program -p firmware.xclbin" hangs? If yes, maybe @stsoe has some insight?

Yes, that is what I'm saying.

maxzhen, 2019-08-22:
The xclbin download seems to finish successfully. The configure cmd seems hang?

dsorber, 2019-08-22:
> The xclbin download seems to finish successfully. The configure cmd seems hang?

Yes, the command seems to hang and doesn't return. I've tried multiple times. One of those times I give it 10+ minutes.

dsorber, 2019-09-19:
The 2.2.2250 XRT release finally fixes this issue.

closed 2019-09-19 by dsorber