Xilinx/XRT #1957: XRT 2.2.2173 does not support Ubuntu 4.15.0-60

dsorber, 2019-09-03:
I just installed regular OS updates, which included kernel 4.15.0-60, on my Ubuntu machine and now XRT won't build:

```
Tue Sep  3 13:08:24 EDT 2019
cd userpf; make all
make[1]: Entering directory '/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf'
echo /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf
make -C /lib/modules/4.15.0-60-generic/build M=/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf modules
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-60-generic'
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_subdev.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_ctx.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../xocl_thread.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/xdma.o
  CC [M]  /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.o
/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c:1686:12: error: static declaration of ‘stream_open’ follows non-static declaration
 static int stream_open(struct inode *inode, struct file *file)
            ^
In file included from ./include/linux/seq_file.h:11:0,
                 from ./include/linux/pinctrl/consumer.h:17,
                 from ./include/linux/pinctrl/devinfo.h:21,
                 from ./include/linux/device.h:24,
                 from ./include/linux/cdev.h:8,
                 from ./include/drm/drmP.h:36,
                 from /var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.c:23:
./include/linux/fs.h:2988:12: note: previous declaration of ‘stream_open’ was here
 extern int stream_open(struct inode * inode, struct file * filp);
            ^
scripts/Makefile.build:330: recipe for target '/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.o' failed
make[3]: *** [/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf/../subdev/qdma.o] Error 1
Makefile:1571: recipe for target '_module_/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf' failed
make[2]: *** [_module_/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-60-generic'
Makefile:54: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/var/lib/dkms/xrt/2.2.2173/build/driver/xclng/drm/xocl/userpf'
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2
```
Everything worked fine on the 4.15.0-58 kernel.  Please advise.

dsorber, 2019-09-04:
Since it's unclear when this will be officially fixed here is a work around for anyone else who gets stuck:

```
$ sudo sed -i '1686s/^.......//' /usr/src/xrt-2.2.2173/driver/xclng/drm/xocl/subdev/qdma.c
$ sudo sed -i '1702s/^.......//' /usr/src/xrt-2.2.2173/driver/xclng/drm/xocl/subdev/qdma.c
$ sudo dkms build -m xrt -v 2.2.2173
$ sudo dkms install -m xrt -v 2.2.2173
```

maxzhen, 2019-09-04:
fix is available on latest 2019.1 branch.

dsorber, 2019-09-04:
@maxzhen when will the fix be released?

maxzhen, 2019-09-04:
maybe in a week or two.

vitesse-ftian, 2019-09-18:
I need 2018.3 version, so following the fix on str_dqma.c (remove static before stream_open/close functions), I can build and install 2.1.1794.   

After sudo /opt/xilinx/xrt/bin/xbutil flash -a all and code boot, I cannot find the card -- sudo lspci -vd 10ee: shows nothing.   Ouch ...

Any advice on what to do?

maxzhen, 2019-09-19:
is driver loaded properly? anything interesting in dmesg log?

dsorber, 2019-09-19:
The 2.2.2250 XRT release finally fixes this issue.

closed 2019-09-19 by dsorber