Xilinx/XRT #2145: XRT error/warning messages printed to stderr

dsorber, 2019-09-25:
The XRT runtime is printing error/warning messages to stderr. These messages disrupt my application.

For example:

```
XRT build version: 2.2.2173
Build hash: ff6624c6d72c6ca68d6b123c9d76e49e55c09097
Build date: 2019-06-27 00:37:59
Git branch: 2019.1
PID: 21275
UID: 1000
[Wed Sep 25 11:59:49 2019]
HOST: r740
EXE: /usr/bin/foo
[XRT] WARNING: Argument '16' of kernel 'krnl_foobar_layer0' is allocated in memory bank 'bank3'; compute unit 'krnl_foobar_layer0_2' cannot be used with this argument and is ignored.
[XRT] WARNING: Argument '16' of kernel 'krnl_foobar_layer0' is allocated in memory bank 'bank2'; compute unit 'krnl_foobar_layer0_1' cannot be used with this argument and is ignored.
```
Is there some mechanism to suppress these messages?

sonals, 2019-10-04:
Add a xrt.ini to current working directory.

```
[Runtime]
  runtime_log=null
```

dsorber, 2019-10-08:
Is there some way to do this globally? I have no idea where someone might run my application from.

dsorber, 2019-11-07:
@sonals Anyone?

stsoe, 2019-11-07:
@dsorber.  There is not any global way to disable other than using xrt.ini to lower verbosity or set the runtime_log to null.   

The warning indicates that some of the kernel's compute units will be unused.  Is that really what you intend?   

When the application creates the cl_kernel it could instead create the kernel with just the compute units that will work with the banks.   For example, XRT supports specifying the kernel names when calling clCreateKernel, e.g.  

> cl_kernel krnl = clCreateKernel(program,"krnl_foobar_layer0:{krnl_foobar_layer0_3}",&err);

The syntax is for the kernel name is:

> "kernel_name:{[cuname]+}" 

where the cu list is comma separated.

dsorber, 2019-11-12:
@stsoe Thanks for the info re kernel name.

I guess the bigger issue is that my code is compiled into a shared library which can then be linked into an application I did not create.  As such XRT writing to stdout/stderr can (and will) cause problems.  I think it would make a lot of sense to add some mechanism to globally disable logging to stdout/stderr because, as I said, I have no control over where someone may run an application that links to my library.  Really, an even better solution would be to create some sort of callback interface so that I can tie XRT logging into my libraries existing logging mechanism.  

Thoughts?

uday610, 2021-07-30:
latest XRT release provides xrt::ini  class in the path: ./include/experimental/xrt_ini.h , please see the example here: https://github.com/Xilinx/XRT/blob/master/tests/xrt/00_hello/main.cpp#L97

closed 2021-07-30 by uday610