Xilinx/XRT #2414: PLATFORM_NOT_FOUND_KHR when running pyopencl tests

Felkin1, 2019-11-07:
I am unable to detect an emulated device when running the pyopencl library and calling the get_platforms() method. I am fully able to run regular host.exe files generated using xocc from the command line when using sdaccel. 

I am usind SDACCEL 2018.2 for building. The toolchain is installed in a different folder than opt, but I took care of this by setting the right env vars. Where is the library looking for the xilinx platform files?

Felkin1, 2019-11-08:
I managed to get it fixed by manually adding an /etc/OpenCL/vendors/xilinx.icd file and changing it to point to the actual implementation in sdaccel's folder.

I have run into another issue now, though, where the python script doesn't have permission to access /usr/bin/.run and loses the env_vars when moving to sudo.

uday610, 2021-07-30:
pyopencl is no longer supported.

closed 2021-07-30 by uday610