Xilinx/XRT #2419: CMake Issues on Ubutnu 19.10

ssbanerje, 2019-11-08:
I am having some difficulties with the CMake build on Ubuntu 19.10 using the current HEAD as well as the 2019.2_RC1 tag. 

When I run `build.sh -driver` the cmake generation script fails to create targets for the tests saying it could not find `OpenSSL::Crypto`. However, cmake spits out an earlier log that it was able to find `OpenSSL`.

Here is a full output of the cmake command.

```
cmake -DRDI_CCACHE=0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON ../../src
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- checking for module 'libdrm'
--   found libdrm, version 2.4.99
-- Looking for DRM - found at /usr 2.4.99
-- checking for module 'OpenCL'
--   found OpenCL, version 2.1
-- Looking for OPENCL - found at /usr 2.1 /usr/include
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Looking for GIT - found at /usr/bin/git
-- Boost version: 1.67.0
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Found Curses: /usr/lib/x86_64-linux-gnu/libcurses.so  
-- XRT CL extension header files
-- include/1_2/CL/cl_ext_xilinx.h
-- include/1_2/CL/cl_ext.h
-- XRT EA eula files  /home/subho/Workshop/XRT/src/../LICENSE
-- Platform/Linux (Ubuntu) (Kernel 5.3.0-19-generic)
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libssl.so;/usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1c")  <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a  
-- Copying test data
-- GTest include dirs: '/usr/include'
-- GTest libraries: '/usr/lib/x86_64-linux-gnu/libboost_system.a;/usr/lib/x86_64-linux-gnu/libboost_filesystem.a;/usr/lib/x86_64-linux-gnu/libboost_program_options.a /usr/lib/x86_64-linux-gnu/libgtest.a;/usr/lib/x86_64-linux-gnu/libgtest_main.a pthread'
-- Configuring CMake to run unit test after building
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
-- XRT header files
-- ert.h
-- stream.h
-- xcl_axi_checker_codes.h
-- xclbin.h
-- xclerr.h
-- xclfeatures.h
-- xclhal2.h
-- xrt.h
-- xcl_app_debug.h
-- xcl_macros.h
-- xcl_app_debug.h
-- xclperf.h
-- xclhal2_mem.h
-- xrt_mem.h
-- XRT experimental header files
-- xrt-next.h
-- XRT header files
-- mpd_plugin.h
-- msd_plugin.h
-- XRT header files
-- mailbox_proto.h
-- Found PROTOBUF: /usr/lib/x86_64-linux-gnu/libprotobuf.so  
-- Found LibXml2: /usr/lib/x86_64-linux-gnu/libxml2.so (found suitable version "2.9.4", minimum required is "2.9.1") 
-- XRT version: 2.3.0
-- Debug DEB package
-- run-clang-tidy found, static code analysis enabled
-- XRT DRIVER SRC BASE DIR /home/subho/Workshop/XRT/src/runtime_src/core
-- XRT DRIVER SRC BASE DIR /home/subho/Workshop/XRT/src/runtime_src/core
-- Preparing OpenCL ICD xilinx.icd
-- Preparing XRT pkg-config
CMake Warning at CMake/coverity.cmake:5 (message):
  -- coverity not found
Call Stack (most recent call first):
  CMake/nativeLnx.cmake:161 (include)
  CMakeLists.txt:61 (include)

-- Configuring done
CMake Error at runtime_src/tools/xclbin/CMakeLists.txt:205 (add_executable):
  Target "xclbintest" links to target "OpenSSL::Crypto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at runtime_src/tools/xclbin/CMakeLists.txt:125 (add_executable):
  Target "xclbinutil" links to target "OpenSSL::Crypto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at runtime_src/tools/xclbin/CMakeLists.txt:125 (add_executable):
  Target "xclbinutil" links to target "OpenSSL::Crypto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/subho/Workshop/XRT/build/Debug

real	0m2.182s
user	0m0.978s
sys	0m0.402s
```

I have installed the dependencies from `src/runtime_src/tools/scripts/xrtdeps.sh`.

ssbanerje, 2019-11-08:
I am not sure if this is related. When I try to build the main libs/bins/kmodules in the Release/Debug directories, I get a lot of errors triggered by the `-Werrror` flags. For example

```
Scanning dependencies of target core_common_objects
[  2%] Building CXX object runtime_src/core/common/CMakeFiles/core_common_objects.dir/scheduler.cpp.o
In file included from /usr/include/string.h:494,
                 from /usr/include/c++/9/cstring:42,
                 from /home/subho/Workshop/XRT/src/runtime_src/core/common/scheduler.cpp:26:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘int xrt_core::scheduler::init(xclDeviceHandle, const axlf*)’ at /home/subho/Workshop/XRT/src/runtime_src/core/common/scheduler.cpp:178:14:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 32 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [runtime_src/core/common/CMakeFiles/core_common_objects.dir/build.make:63: runtime_src/core/common/CMakeFiles/core_common_objects.dir/scheduler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1038: runtime_src/core/common/CMakeFiles/core_common_objects.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
```
```
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h: In instantiation of ‘std::vector<xrt::device> xrt::loadDevices(UnaryPredicate) [with UnaryPredicate = xrt::loadDevices()::<lambda(const string&)>]’:
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:982:26:   required from here
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:968:27: error: moving a local object in a return statement prevents copy elision [-Werror=pessimizing-move]
  968 |   return std::move(devices);
      |                           ^
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:968:27: note: remove ‘std::move’ call
In file included from /home/subho/Workshop/XRT/src/runtime_src/xocl/core/device.h:25,
                 from /home/subho/Workshop/XRT/src/runtime_src/xdp/profile/plugin/ocl/xocl_profile.h:29,
                 from /home/subho/Workshop/XRT/src/runtime_src/xdp/profile/plugin/ocl/xocl_profile.cpp:17:
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h: In instantiation of ‘std::vector<xrt::device> xrt::loadDevices(UnaryPredicate) [with UnaryPredicate = xrt::loadDevices()::<lambda(const string&)>]’:
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:982:26:   required from here
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:968:27: error: moving a local object in a return statement prevents copy elision [-Werror=pessimizing-move]
  968 |   return std::move(devices);
      |                           ^
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:968:27: note: remove ‘std::move’ call
In file included from /home/subho/Workshop/XRT/src/runtime_src/xocl/core/device.h:25,
                 from /home/subho/Workshop/XRT/src/runtime_src/xdp/profile/plugin/ocl/xocl_profile.h:29,
                 from /home/subho/Workshop/XRT/src/runtime_src/xdp/profile/plugin/ocl/xocl_plugin.h:21,
                 from /home/subho/Workshop/XRT/src/runtime_src/xdp/profile/plugin/ocl/xocl_plugin.cpp:17:
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h: In instantiation of ‘std::vector<xrt::device> xrt::loadDevices(UnaryPredicate) [with UnaryPredicate = xrt::loadDevices()::<lambda(const string&)>]’:
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:982:26:   required from here
/home/subho/Workshop/XRT/src/runtime_src/xrt/device/device.h:968:27: error: moving a local object in a return statement prevents copy elision [-Werror=pessimizing-move]
  968 |   return std::move(devices);
```

stsoe, 2019-11-08:
@ssbanerje the compilation errors are code problems indeed.   Not yet sure about the CMake errors.  We will check it out and fix.  Thanks for pointing out the problems.

eejlny, 2019-12-06:
Hello, 

Any updates on this ?I am also seeing the same problem when trying to run build.sh

CMake Error at runtime_src/tools/xclbin/CMakeLists.txt:125 (add_executable):
  Target "xclbinutil" links to target "OpenSSL::Crypto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

foolmarks, 2019-12-15:
I'm seeing the same error when building on Ubuntu 18.03 LTS.

stsoe, 2019-12-15:
@foolmarks Do you mean 18.04.3 ?  We are building and releasing packages on 18.04 by default.
While Ubuntu 19.10 has compile errors with gcc9.2, Ubuntu 18.04 is clean.
Did you install the necessary packages by running xrtdeps.sh prior to building XRT?
Which branch are you building?

@ssbanerje @eejlny I will fix the gcc9.2 compile errors, but please beware that XRT doesn't officially support Ubuntu 19.10, alas it is not built or tested on that platform.  Any fixes for gcc9.2 will be on master only.

Lastly, if you have run xrtdpes.sh and are still seeing CMake errors, then maybe there are stale CMake generated files, please try ./build.sh -clean and then build again.

foolmarks, 2019-12-15:
Sorry, yes I did mean 18.04.3 LTS

Yes, I've run the dependency install script (several times).

I'm building the master branch.

stsoe, 2019-12-15:
@foolmarks That's odd.  I am building the master branch on 18.04.3 no problems, and you cleaned stale CMake data with 

> build.sh -clean

 first?

foolmarks, 2019-12-16:
Yes, I did the clean (several times).

Just tried the build of the master branch on 18.04.2 and it ran perfectly - I ended up with several .deb packages in the Release folder:

xrt_202010.2.5.0_18.04-aws.deb
xrt_202010.2.5.0_18.04-azure.deb
xrt_202010.2.5.0_18.04-container.deb
xrt_202010.2.5.0_18.04-xbtest.deb
xrt_202010.2.5.0_18.04-xrt.deb

I'm guessing that I need to install  xrt_202010.2.5.0_18.04-xrt.deb on my local machine.

I haven't made any packages in the Debug folder - do I need them?

Also, I'm a little unsure what exactly 'please copy aside /lib/firmware/xilinx/sched*.bin' means.   Should I make a temporary copy of all /lib/firmware/xilinx/sched*.bin files, then install the .deb and then copy the original sched*.bin back to where they were?

stsoe, 2019-12-16:
Yes, if you build XRT yourself, then before installing your locally built package xrt_202010.2.5.0_18.04-xrt.deb, copy /lib/firmware/xilinx/sched*.bin to a temporary location.  After installing your locally built package, you will see that apt has removed  /lib/firmware/xilinx/sched*.bin, now simply copy them back.  We are working on a better solution for this, e.g. decoupling the scheduler firmware from the XRT package.

I am still not understanding the problem you run into when building on 18.04.3.  I created a docker container from scratch with 18.04.3, ran xrtdeps.sh,  and was able to build XRT master no problem.

foolmarks, 2019-12-16:
I got this to work by doing the following:
- installed 18.04.2 from scratch. Allowed it to update to 18.04.3
- cloned the github repo, ran the dependencies script then built the .deb
- installed the XRT .deb from build/Release   ( I don't have any other Xilinx tools installed so didn't need to save the  /lib/firmware/xilinx/sched*.bin files).
- downloaded and installed the xilinx-u200-xdma-201830.2-2580015_18.04.deb deployment shell.

After that, I was able to flash the Alveo U200 card.

stsoe, 2019-12-16:
Without the sched*.bin files you will not be able to use XRT with the embedded scheduler.
I suggest you download the XRT package, extract (dpkg -x) it some some directory, then copy extracted sched*.bin files  to /lib/firmware/xilinx.

stsoe, 2019-12-16:
Closing as fixed with PR #2595

closed 2019-12-16 by stsoe