Xilinx/XRT #2565: zocl_drv.c:421:8: error: unknown type name 'vm_fault_t'

Vae1997, 2019-12-10:
I am referencing: https://github.com/Xilinx/XRT/blob/master/src/runtime_src/doc/toc/yocto.rst

But when I execute petalinux-build, I get compilation errors:

Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 KERNEL_SRC=/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source KERNEL_PATH=/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source KERNEL_VERSION=4.14.0-xilinx-v2018.3 CC=aarch64-xilinx-linux-gcc   -fuse-ld=bfd LD=aarch64-xilinx-linux-ld.bfd   AR=aarch64-xilinx-linux-ar  O=/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make -C /home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source M=/home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl modules
| make[1]: Entering directory '/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
| make[2]: Entering directory '/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-build-artifacts'
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/sched_exec.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_sysfs.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_ert.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_ioctl.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_drv.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_bo.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_dma.o
|   CC [M]  /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_cu.o
| /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_drv.c:421:8: error: unknown type name 'vm_fault_t'
|  static vm_fault_t zocl_bo_fault(struct vm_fault *vmf)
|         ^~~~~~~~~~
| make[3]: *** [/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source/scripts/Makefile.build:314: /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl/zocl_drv.o] Error 1
| make[3]: *** Waiting for unfinished jobs....
| make[2]: *** [/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source/Makefile:1503: _module_/home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/git/src/runtime_src/core/edge/drm/zocl] Error 2
| make[2]: Leaving directory '/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-build-artifacts'
| make[1]: *** [Makefile:146: sub-make] Error 2
| make[1]: Leaving directory '/home/wzy/zh-project/BSP_project/build/tmp/work-shared/zcu102-zynqmp/kernel-source'
| make: *** [Makefile:25: modules] Error 2
| ERROR: oe_runmake failed
| WARNING: /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/temp/run.do_compile.7710:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/wzy/zh-project/BSP_project/build/tmp/work/zcu102_zynqmp-xilinx-linux/zocl/2.2.0+gitAUTOINC+3b1999f9b3-r0/temp/log.do_compile.7710)
ERROR: Task (/home/wzy/zh-project/BSP_project/project-spec/meta-user/recipes-xrt/zocl/zocl_git.bb:do_compile) failed with exit code '1'

what should I do? Looking forward to your reply!

larry9523, 2019-12-10:
Looks like you are using 'KERNEL_VERSION=4.14.0-xilinx-v2018.3 ', there is no vm_fault_t before 4.17. Which XRT branch are you using? Can you update your petalinux or switch to 2018.3 XRT?

Vae1997, 2019-12-10:
> Looks like you are using 'KERNEL_VERSION=4.14.0-xilinx-v2018.3 ', there is no vm_fault_t before 4.17. Which XRT branch are you using? Can you update your petalinux or switch to 2018.3 XRT?

Yes, I am using 4.14.0-xilinx-v2018.3, but the latest branch for XRT, so which XRT branch should I use?

Vae1997, 2019-12-10:
OK,now it works well.
I just changed the XRT to 2018.3,thanks a lot.😊

closed 2019-12-10 by Vae1997