Xilinx/XRT #2610: Build issue on RHEL 7.5

singagan, 2019-12-19:
Hi, 
After running src/runtime_src/tools/scripts/xrtdeps.sh on RHEL 7.5 when I try to build XRT I get the error that cmake generation script fails saying it could not find OpenSSL::Crypto

I run enable devtools prior to running build.sh.

This is similar to #2419  
So I have tried to follow the suggestions, however, I am still having the issue on rhel7.5.

Below is the log:
cmake -DRDI_CCACHE=0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON ../../src
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Found PkgConfig: /bin/pkg-config (found version "0.27.1")
-- checking for module 'libdrm'
--   found libdrm, version 2.4.83
-- Looking for DRM - found at /usr 2.4.83
-- checking for module 'OpenCL'
--   found OpenCL, version 2.2
-- Looking for OPENCL - found at /usr 2.2 /usr/include
-- Found Git: /bin/git (found version "1.8.3.1")
-- Looking for GIT - found at /bin/git
-- Boost version: 1.58.0
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
-- Found Curses: /lib64/libcurses.so
-- XRT CL extension header files
-- include/1_2/CL/cl_ext_xilinx.h
-- include/1_2/CL/cl_ext.h
-- XRT EA eula files  /dataL/sin/XRT/src/../LICENSE
-- Platform/Linux (RedHatEnterpriseWorkstation) (Kernel 3.10.0-862.el7.x86_64)
-- Compiler: /bin/c++ /bin/cc
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Found OpenSSL: /usr/lib64/libssl.so;/usr/lib64/libcrypto.so (found version "1.0.2k")
-- Found ZLIB: /dataL/sin/anaconda3/lib/libz.so (found version "1.2.11")
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Found GTest: /lib64/libgtest.so
-- Copying test data
-- GTest include dirs: '/usr/include'
-- GTest libraries: '/usr/local/lib/libboost_system.a;/usr/local/lib/libboost_filesystem.a;/usr/local/lib/libboost_program_options.a /lib64/libgtest.so;/lib64/libgtest_main       .so pthread'
-- Configuring CMake to run unit test after building
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
CMake Warning at runtime_src/ert/CMakeLists.txt:43 (message):
  ****************************************************************

  No firmware files built or copied, resulting XRT package will be missing
  ERT scheduler firmware.  Use build.sh -ertfw <dir> to specify path to a
  directory with firmware to copy during XRT build.

  ****************************************************************

-- XRT header files
-- ert.h
-- xstream.h
-- types.h
-- xcl_axi_checker_codes.h
-- xclbin.h
-- xclerr.h
-- xclfeatures.h
-- xclhal2.h
-- xrt.h
-- xcl_app_debug.h
-- xcl_macros.h
-- xcl_app_debug.h
-- xclperf.h
-- xclhal2_mem.h
-- xrt_mem.h
-- XRT experimental header files
-- xrt-next.h
-- XRT OS native header files
-- types.h
-- uuid.h
bfl=/usr/local/lib/libboost_filesystem.so
-- XRT header files
-- mpd_plugin.h
-- msd_plugin.h
-- XRT header files
-- mailbox_proto.h
-- Found PROTOBUF: /usr/local/lib/libprotobuf.so
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Found LibXml2: /usr/lib64/libxml2.so (found suitable version "2.9.1", minimum required is "2.9.1")
-- checking for modules 'json-glib-1.0;glib-2.0'
--   found json-glib-1.0, version 1.2.6
--   found glib-2.0, version 2.56.1
-- XRT version: 2.5.0
-- Debug RPM package
CMake Warning at CMake/lint.cmake:10 (message):
  -- run-clang-tidy not found, static code analysis disabled
Call Stack (most recent call first):
  CMake/nativeLnx.cmake:145 (include)
  CMakeLists.txt:62 (include)

-- XRT DRIVER SRC BASE DIR /dataL/sin/XRT/src/runtime_src/core
-- XRT DRIVER SRC BASE DIR /dataL/sin/XRT/src/runtime_src/core
-- Preparing OpenCL ICD xilinx.icd
-- Preparing XRT pkg-config
CMake Warning at CMake/coverity.cmake:5 (message):
  -- coverity not found
Call Stack (most recent call first):
  CMake/nativeLnx.cmake:164 (include)
  CMakeLists.txt:62 (include)

-- Configuring done
CMake Error at runtime_src/tools/xclbin/CMakeLists.txt:213 (add_executable):
  Target "xclbintest" links to target "OpenSSL::Crypto" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

stsoe, 2019-12-19:
@singagan Please follow the exact instructions in the documentation.  For example, you are not building using the right software collection.  'scl enable devtoolset-6 bash' .  Do this and run build.sh clean before trying again.

helmotz, 2020-04-03:
@stsoe @singagan I recently had the same issue trying to build tag 2019.2 on my RHEL 7.6 workstation.
Apparently cmake version 3.3.2 doesn't supply the OpenSSL::Crypto target (I had to change it with ${OPENSSL_CRYPTO_LIBRARY}). 
Moreover it looks like the -Werror compiler option used in src/runtime_src/CMakeLists.txt is too strict and causes compilation errors.
Please find attached the changes that I applied.

[patch.diff.gz](https://github.com/Xilinx/XRT/files/4425294/patch.diff.gz)

uday610, 2021-07-30:
Not supported

closed 2021-07-30 by uday610