Xilinx/XRT #2632: AWS F1 install XRT, report xclmgmt is missing

LeonMac, 2020-01-04:
Other step are running OK, when doing deb package installation, always meet issue as below:
$ sudo apt install --reinstall ./xrt_202010.2.5.0_18.04-xrt.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'xrt' instead of './xrt_202010.2.5.0_18.04-xrt.deb'
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/ubuntu/aws-fpga/SDAccel/Runtime/XRT_2019.1/build/Debug/xrt_202010.2.5.0_18.04-xrt.deb xrt amd64 2.5.0 [44.2 MB]
(Reading database ... 132438 files and directories currently installed.)
Preparing to unpack .../xrt_202010.2.5.0_18.04-xrt.deb ...
Unregistering old XRT Linux kernel module sources 2.5.0 from dkms on Ubuntu/Debian

-------- Uninstall Beginning --------
Module:  xrt
Version: 2.5.0
Kernel:  4.15.0-1056-aws (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

xocl.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.15.0-1056-aws/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

xclmgmt.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.15.0-1056-aws/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.5.0
completely from the DKMS tree.
------------------------------
Done.
Cleanup is skipped for package upgrade/downgrade/re-install on Ubuntu/Debian
Unpacking xrt (2.5.0) over (2.5.0) ...
Setting up xrt (2.5.0) ...
Stopping MPD daemon
Unloading old XRT Linux kernel modules
rmmod: ERROR: Module xclmgmt is not currently loaded
Invoking DKMS common.postinst for xrt
Loading new xrt-2.5.0 DKMS files...
Building for 4.15.0-1056-aws
Building initial module for 4.15.0-1056-aws
Done.

xocl:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.15.0-1056-aws/updates/dkms/

xclmgmt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.15.0-1056-aws/updates/dkms/

depmod...

DKMS: install completed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see dmesg)
Installing MSD / MPD daemons
Starting MPD daemon
Installing pyopencl...
....
_________________
$ dmesg | grep xclmgmt
[  699.257353] xclmgmt: Unknown symbol fpga_mgr_unregister (err 0)
[  699.257397] xclmgmt: Unknown symbol fpga_mgr_register (err 0)

LeonMac, 2020-01-05:
I were told the xclmgmt will NOT work on AWS F1 machine (AWS customize the PF MGMT).
Please comments--if that is true I will close the issue

LeonMac, 2020-01-09:
Issue solved by @xuhz , not XRT relevant. Thanks so much!

closed 2020-01-09 by LeonMac

sqPoseidon, 2021-09-27:
> Issue solved by @xuhz , not XRT relevant. Thanks so much!

Hi,

I met the same problem on my local host: `5.4.0-62-generic #70~18.04.1-Ubuntu SMP Tue Jan 12 17:18:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`. 

```shell
(base) xxx@ThinkCentre-M720s:~/xxxxxx$ sudo apt install ./xrt_202020.2.8.743_18.04-amd64-xrt.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'xrt' instead of './xrt_202020.2.8.743_18.04-amd64-xrt.deb'
The following NEW packages will be installed:
  xrt
0 upgraded, 1 newly installed, 0 to remove and 94 not upgraded.
Need to get 0 B/8647 kB of archives.
After this operation, 50.5 MB of additional disk space will be used.
Get:1 /home/xxx/xxxxxx/xrt_202020.2.8.743_18.04-amd64-xrt.deb xrt amd64 2.8.743 [8647 kB]
Selecting previously unselected package xrt.
(Reading database ... 352816 files and directories currently installed.)
Preparing to unpack .../xrt_202020.2.8.743_18.04-amd64-xrt.deb ...
Unpacking xrt (2.8.743) ...
Setting up xrt (2.8.743) ...
Unloading old XRT Linux kernel modules
rmmod: ERROR: Module xocl is not currently loaded
rmmod: ERROR: Module xclmgmt is not currently loaded
Invoking DKMS common.postinst for xrt
Loading new xrt-2.8.743 DKMS files...
Building for 5.4.0-62-generic
Building initial module for 5.4.0-62-generic
Secure Boot not enabled on this system.
Done.

xocl:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-62-generic/updates/dkms/

xclmgmt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-62-generic/updates/dkms/

depmod...

DKMS: install completed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see dmesg)
Installing MSD / MPD daemons
(base) xxx@xxx-ThinkCentre-M720s:~/xxxxxx$
```

How did you solve this problem?

Thank you in advance.