Xilinx/XRT #2636: Install XRT onto AWS F1 baremetal but xbutil scan fail.

LeonMac, 2020-01-05:
Hi I am trying to install XRT onto one AWS F1 baremetal machine, but meet issue that xbutil scan fail.
Below are the logs:
_____________________
$xbutil scan
INFO: Found total 1 card(s), 0 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.15.0-1056-aws
Version:	#58-Ubuntu SMP Tue Nov 26 15:14:34 UTC 2019
Machine:	x86_64
Model:		HVM domU
CPU cores:	8
Memory:		122887 MB
Glibc:		2.27
Distribution:	Ubuntu 18.04.3 LTS
Now:		Sun Jan  5 15:49:14 2020
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.5.0
Git Hash:	
Git Branch:	
Build Date:	2020-01-03 11:17:41
XOCL:		2.5.0,
XCLMGMT:	unknown
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*[0] 0000:00:1d.0 xilinx_aws-vu9p-f1_dynamic_5_0(ts=0xabcd) user(inst=128)
WARNING: card(s) marked by '*' are not ready, is MPD runing? run 'systemctl status mpd' to check MPD details.
$ xbutil query
INFO: Found total 1 card(s), 0 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.15.0-1056-aws
Version:	#58-Ubuntu SMP Tue Nov 26 15:14:34 UTC 2019
Machine:	x86_64
Model:		HVM domU
CPU cores:	8
Memory:		122887 MB
Glibc:		2.27
Distribution:	Ubuntu 18.04.3 LTS
Now:		Sun Jan  5 15:49:26 2020
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.5.0
Git Hash:	
Git Branch:	
Build Date:	2020-01-03 11:17:41
XOCL:		2.5.0,
XCLMGMT:	unknown
ERROR: Card index 0 is out of range
$ systemctl status mpd.service
● mpd.service - Xilinx Management Proxy Daemon (MPD)
   Loaded: loaded (/etc/systemd/system/mpd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-01-04 08:17:08 UTC; 1 day 7h ago
 Main PID: 17775 (mpd)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/mpd.service
           └─17775 /opt/xilinx/xrt/bin/mpd

Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: create thread pair for 0000:00:1d.0
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: 1 pairs of threads running...
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] write 56 bytes out of 56 bytes to fd 4
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] msg arrived on mailbox fd 4
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] retrieved msg size from mailbox: 40 bytes
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] read 72 bytes out of 72 bytes from fd 4, valid: 1
Jan 04 08:17:08 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] mpd daemon: request 11 received(reqSize: 24)
Jan 04 08:17:11 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] msg arrived on mailbox fd 4
Jan 04 08:17:11 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] retrieved msg size from mailbox: 32 bytes
Jan 04 08:17:11 ip-172-31-25-76 mpd[17775]: [0:0:1d.0] read 64 bytes out of 64 bytes from fd 4, valid: 1

LeonMac, 2020-01-09:
Issue solved by @xuhz , not XRT relevant. 
Thanks so much!

closed 2020-01-09 by LeonMac

rDaffa, 2020-01-28:
I am having the same issue. can @LeonMac or @xuhz share with us the solution?

LeonMac, 2020-01-28:
@rDaffa My issue are mixture of several kinds of reason so hardly to simply give a directly solution
Can you be more spcify the detail--if you look into the message hostory, there are some tips suggested. maybe you could paste more infor. about problem you meet.