Xilinx/XRT #2686: undefined reference to `boost::system::detail::generic_category_instance'

arrow53, 2020-01-16:
On Ubuntu 18.04 with either master or 2019.2 I get this error

```
../../../common/libxrt_coreutil_static.a(config_reader.cpp.o): In function `boost::system::generic_category()':
/usr/local/include/boost/system/error_code.hpp:478: undefined reference to `boost::system::detail::generic_category_instance'
collect2: error: ld returned 1 exit status
make[2]: *** [runtime_src/core/pcie/tools/awssak/CMakeFiles/awssak.dir/build.make:86: runtime_src/core/pcie/tools/awssak/awssak] Error 1
make[1]: *** [CMakeFiles/Makefile2:1588: runtime_src/core/pcie/tools/awssak/CMakeFiles/awssak.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

stsoe, 2020-01-18:
Hi @arrow53.  This is strange since we build XRT on Ubuntu 18.04 with no problems.  Can you provide more insight into how you are building?  Do you maybe have your own version of boost installed?  I see other reports of same error when I google.

arrow53, 2020-01-18:
I can build fine on another Ubuntu 18.04 fine too. I'll just close it. I don't have my own boost installed that I know of and did try to purge/reinstall boost system/filesystem with no joy. 
The deb package here did work fine if anyone else runs across this and just wants to use xrt.

https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/Chunk1858803630.html

closed 2020-01-18 by arrow53