Xilinx/XRT #2728: Cannot install on CentOS 7.5

jrwagz, 2020-01-28:
When trying to install the published RPM for CentOS on CentOS 7.5 I get the following errors:

```
[root@cnes152 alveo]# yum install xrt_201920.2.3.1301_7.4.1708-xrt.rpm
Loaded plugins: fastestmirror
Examining xrt_201920.2.3.1301_7.4.1708-xrt.rpm: xrt-2.3.1301-1.x86_64
Marking xrt_201920.2.3.1301_7.4.1708-xrt.rpm to be installed
Resolving Dependencies
--> Running transaction check
---> Package xrt.x86_64 0:2.3.1301-1 will be installed
--> Finished Dependency Resolution
 
Dependencies Resolved
 
===================================================================================================================
Package          Arch                Version                 Repository                                      Size
===================================================================================================================
Installing:
xrt              x86_64              2.3.1301-1              /xrt_201920.2.3.1301_7.4.1708-xrt               26 M
 
Transaction Summary
===================================================================================================================
Install  1 Package
 
Total size: 26 M
Installed size: 26 M
Is this ok [y/d/N]: y
Downloading packages:
Running transaction check
Running transaction test
Transaction test succeeded
Running transaction
  Installing : xrt-2.3.1301-1.x86_64                                                                           1/1
Unloading old XRT Linux kernel modules
rmmod: ERROR: Module xocl is not currently loaded
rmmod: ERROR: Module xclmgmt is not currently loaded
Invoking DKMS common.postinst for xrt
Loading new xrt-2.3.1301 DKMS files...
Building for 3.10.0-862.el7.x86_64
Building initial module for 3.10.0-862.el7.x86_64
Error!  Build of xocl.ko failed for: 3.10.0-862.el7.x86_64 (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/xrt/2.3.1301/build/ for more information.
****************************************************************
* DKMS failed to install XRT drivers.
* Please check if kernel development headers are installed for OS variant used.
*
* Check build logs in /var/lib/dkms/xrt/2.3.1301
***************************************************************
```

It appears that xocl is failing to build, and here is the detailed error from it's make.log:

```
DKMS make.log for xrt-2.3.1301 for kernel 3.10.0-862.el7.x86_64 (x86_64)
Mon Jan 27 17:01:13 MST 2020
cd userpf; make all
make[1]: Entering directory `/var/lib/dkms/xrt/2.3.1301/build/driver/xocl/userpf'
echo /var/lib/dkms/xrt/2.3.1301/build/driver/xocl/userpf
/var/lib/dkms/xrt/2.3.1301/build/driver/xocl/userpf
make -C /lib/modules/3.10.0-862.el7.x86_64/build M=/var/lib/dkms/xrt/2.3.1301/build/driver/xocl/userpf modules
make[2]: Entering directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
arch/x86/Makefile:96: stack-protector enabled but compiler support broken
arch/x86/Makefile:166: *** CONFIG_RETPOLINE=y, but not supported by the compiler. Compiler update recommended..  Stop.
make[2]: Leaving directory `/usr/src/kernels/3.10.0-862.el7.x86_64'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/var/lib/dkms/xrt/2.3.1301/build/driver/xocl/userpf'
make: *** [all] Error 2
```

zohourih, 2020-03-12:
Do you have GCC installed? If yes, then you need to use a newer version of GCC than what you have installed.

uday610, 2021-07-30:
it should be working now. 7.5 is no longer officially supported though

closed 2021-07-30 by uday610