Xilinx/XRT #2841: How to compile xrt for Edge?

niclasb, 2020-02-14:
How is xrt for Edge supposed to be compiled? 

This line disables edge if compiling on target. 
https://github.com/Xilinx/XRT/blob/e68552d741d40c47bf4ac2edeff9f54aa18361f0/src/runtime_src/core/CMakeLists.txt#L7

I tried crosscompile xrt but ran into many issues so I just changed XRT_NATIVE_BUILD to no in the top build and that works fine on target. 

Regards, 
Niclas

zohourih, 2020-02-21:
Actually I am also trying to do something related where I am trying to somehow get XRT working on a Zynq MPSoC board with Ubuntu RootFS. My very last failed attempt was to copy the content of `/lib/modules/` from PetaLinux RootFS to Ubuntu RootFS and load the zocl driver manually using modprobe, which did actually succeed, but even then the `cl::Platform::get()` call fails with error -1001. I guess we are not supposed to run XRT on any OS except PetaLinux for edge platforms, in which case the zocl driver gets integrated into PetaLinux when building it.

niclasb, 2020-02-21:
I have it running to some extent with Ubuntu 18.04. I guess you have some problems with the shared libraries. What does "xbutil scan" say?
And check that you have env variable XILINX_XRT set (/usr works if you used make install).

zohourih, 2020-02-21:
Did you compile XRT on the Ubuntu RootFS running on an SoC board? Or are you trying to cross-compile XRT for an SoC board using Ubuntu running on an x86 machine?

I am trying the former case because I want to use Ubuntu instead of PetaLinux on my SoC board (ZCU104) but so far, I have not been able to successfully compile XRT (on Ubuntu running on the SoC board itself) since I neither know how I should install kernel headers for Xilinx's kernel, nor do I know how to get the XRT compilation script to find the kernel headers in such case. I ended up copying the whole RootFS from PetaLinux with XRT and zocl integrated to the Ubuntu RootFS running on the SoC, copied `/lib/modules/4.19.0-xilinx-v2019.2` from PetaLinux RootFS to Ubuntu RootFS and manually loaded the zocl driver from there, and set XILINX_XRT to point to the `/usr` folder from PetaLinux RootFS, but got the error I mentioned above trying to execute my kernel (which, by the way, runs just fine if I use PetaLinux RootFS instead of Ubuntu).

niclasb, 2020-02-21:
I built xrt on the soc itself.
You will need the kernel headers (make deb-pkg) then install the three deb packages on target. Jump to zocl dir under xrt and run make. Then you will have zocl.ko.
Build xrt and run make install.

zohourih, 2020-02-25:
@niclasb Thank you for the great tip. I finally managed to correctly create DEB packages for Xilinx's kernel and install them. Then I downloaded the "First 2019.2_PU2 release" package, changed `src/CMakeLists.txt`, Line 20 to `set(XRT_NATIVE_BUILD "no")`, ran `make` in `src/runtime_src/core/edge/drm/zocl`, ran `./build.sh -opt` on top directory, and finally ran `make modules_install` in `src/runtime_src/core/edge/drm/zocl` to install the zocl driver. Now, when I run `xbutil scan` I get:

```xbutil scan
INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:        Linux
Release:        4.19.0-xilinx-v2019.2
Version:        #1 SMP Wed Feb 19 08:42:39 UTC 2020
Machine:        aarch64
Glibc:          2.27
Distribution:   Ubuntu 18.04.4 LTS
Now:            Tue Feb 25 17:27:57 2020
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:        2.5.0
Git Hash:
Git Branch:
Build Date:     2020-02-25 16:56:40
ZOCL:           2018.2.1
 [0]:zcu104_base
```

Then I ran `export XILINX_XRT=/usr`. Yet, running my host code still fails with that dreaded `Error calling err = cl::Platform::get(&platforms), error code is: -1001`!!! I can even successfully reconfigure the FPGA using `xbutil program`, but the host code refuses to see the FPGA. In my experience, error -1001 usually happens when the OpenCL icd driver doesn't exist in `/etc/OpenCL/vendors/`. However, in the case of Xilinx MPSoC devices, when PetaLinux is used, the above path doesn't even exist and yet the host code runs fine.

Have you successfully ran OpenCL-based host codes (Vitis flow) on your setup?

niclasb, 2020-02-25:
Try setting /etc/OpenCL/vendors/xilinx.icd to libxilinxopencl.so

Run strace on the application and check if it finds the correct libs.

zohourih, 2020-02-25:
Yes, doing `echo "/usr/lib/libxilinxopencl.so" > /etc/OpenCL/vendors/xilinx.icd` fixes the problem... I should have tried it myself first before posting here; the fact that the icd file doesn't exist on PetaLinux mislead me into thinking the problem must be something else. Again thank you for your great tips and sorry for hijacking your thread.

Now let's hope Xilinx provides a native way to install XRT directly on Edge platforms.

mKuznii, 2020-03-04:
>  You will need the kernel headers (make deb-pkg) then install the three deb packages on target. Jump to zocl dir under xrt and run make. Then you will have zocl.ko.

I am also running Ubuntu 18.04 on zcu104 and trying to install XRT on it. How did you manage to create DEB packages for Xilinx's kernel?

niclasb, 2020-03-04:
Build the kernel with "make deb-pkg" and you will get the deb packages.

zohourih, 2020-03-05:
> I am also running Ubuntu 18.04 on zcu104 and trying to install XRT on it. How did you manage to create DEB packages for Xilinx's kernel?

Clone the xilinx kernel repo from [here](https://github.com/Xilinx/linux-xlnx) on the SoC, checkout the exact version of the kernel that you used to build PetaLinux (e.g. if you used the 2019.2 version of PetaLinux, checkout the 2019.2 branch), copy `.config` and `.config.old` from your PetaLinux build directory to the root folder of the cloned kernel repository, use `make deb-pkg` to get DEB packages.

eejlny, 2020-03-09:
I am trying to do something similar and wondering if it is not possible to use the XRT outputs from the Petalinux RootFS in the UbuntuFS at all ? I was hoping there is a way.

zohourih, 2020-03-10:
@eejlny What do you mean by XRT outputs? Can you elaborate more on what you are trying to do?

eejlny, 2020-03-10:
Basically I wanted to copy zocl.ko and other XRT files generated in the Petalinux rootfs to the Ubuntu Roofts without having to recompile them on the SoC again.

zohourih, 2020-03-10:
@eejlny I already tried that but couldn't succeed in the end; compiling XRT on the SoC with Ubuntu RootFS works, though, and I have already ran OpenCL-based code on this setup after compiling XRT on the SoC. If you follow my earlier replies in this issue, you should be able to do the same.

mKuznii, 2020-03-26:
I am trying to run Vitis AI samples on zcu104 with Ubutnu 18.04 running. I build and install all required packages. XRT was build like it is written above. My output from xbutil scan is:
```
INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.19.0-xilinx-v2019.2
Version:	#1 SMP Sun Mar 22 00:15:07 UTC 2020
Machine:	aarch64
Glibc:		2.27
Distribution:	Ubuntu 18.04.4 LTS
Now:		Thu Mar 26 12:27:22 2020
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.5.0
Git Hash:	
Git Branch:	
Build Date:	2020-03-26 11:44:47
ZOCL:		2018.2.1
 [0]:edge
```

I think this means that my board sees the driver. I also correct problems with running dpuclk.service  so it loads correctly. When I try to run any of the examples from Vitis AI library (let's say classification) I get this error message:
```
Load Xclbin Failed
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0326 12:35:10.314750  3618 xrt_bin_stream.cpp:143] Check failed: xclLoadXclBin(handle, blob) == 0 (-1 vs. 0) Bitstream download failed !
*** Check failure stack trace: ***
Aborted
```
Does anybody know where could be a problem and how to solve it? I am trying to use Vitis AI version 1.0

wilderfield, 2020-05-11:
> Clone the xilinx kernel repo from [here](https://github.com/Xilinx/linux-xlnx) on the SoC, checkout the exact version of the kernel that you used to build PetaLinux (e.g. if you used the 2019.2 [...]

Shouldn't the linux kernel headers be nearly identical to the main line kernel headers for the same kernel version? i.e. 4.19 ? Could users just install the generic 4.19 version?

zohourih, 2020-05-11:
@wilderfield You cannot install the generic Linux _headers_ package in your build because you cannot install the generic kernel package itself and the former package depends on the latter.

wilderfield, 2020-05-12:
> Clone the xilinx kernel repo from [here](https://github.com/Xilinx/linux-xlnx) on the SoC, checkout the exact version of the kernel that you used to build PetaLinux (e.g. if you used the 2019.2 [...]

I was able to do this on an x86 Ubuntu Host by:
```
sudo apt-get install libssl-dev fakeroot
git clone https://github.com/Xilinx/linux-xlnx
cd linux-xlnx
git checkout xlnx_rebase_v4.19_2019.2
(copy the .config here)
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- deb-pkg
```

I am pretty sure I used the right tag, that corresponds to PetaLinux 2019.2
I previously had the linaro gcc and g++ packages apt-get installed

wilderfield, 2020-05-14:
> @niclasb Thank you for the great tip. I finally managed to correctly create DEB packages for Xilinx's kernel and install them (took me some time figure out I need to copy the `.config` file from my PetaLinux build to the kernel source folder, though). Then I downloaded the `First 2019.2_PU2 release` XRT package, changed `src/CMakeLists.txt`, Line 20 to `set(XRT_NATIVE_BUILD "no")`, ran `make` in `src/runtime_src/core/edge/drm/zocl`, ran `./build.sh -opt` on top directory, ran `make install` in `build/Release`, and finally ran `make modules_install` in `src/runtime_src/core/edge/drm/zocl` to install the zocl driver. Now, when I run `xbutil scan` I get:
> 
> ```
> INFO: Found total 1 card(s), 1 are usable
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> System Configuration
> OS name:        Linux
> Release:        4.19.0-xilinx-v2019.2
> Version:        #1 SMP Wed Feb 19 08:42:39 UTC 2020
> Machine:        aarch64
> Glibc:          2.27
> Distribution:   Ubuntu 18.04.4 LTS
> Now:            Tue Feb 25 17:27:57 2020
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> XRT Information
> Version:        2.5.0
> Git Hash:
> Git Branch:
> Build Date:     2020-02-25 16:56:40
> ZOCL:           2018.2.1
>  [0]:zcu104_base
> ```
> 
> This shows that the FPGA is seen correctly by the driver.
> 
> Then I ran `export XILINX_XRT=/usr` and `export LD_LIBRARY_PATH=/usr/lib`. Yet, running my host code still fails with that dreaded `Error calling err = cl::Platform::get(&platforms), error code is: -1001`!!! I can even successfully reconfigure the FPGA using `xbutil program`, but the host code refuses to see the FPGA. In my experience, error -1001 usually happens when the OpenCL icd driver doesn't exist in `/etc/OpenCL/vendors/`. However, in the case of Xilinx MPSoC devices, when PetaLinux is used, the above path doesn't exist at all and yet the host code runs fine; hence, the problem must be something else.
> 
> Have you successfully ran OpenCL-based host codes (Vitis flow) on your setup?

A subtlety here that I ran into, is you must run xbutil as root

zohourih, 2020-05-15:
> A subtlety here that I ran into, is you must run xbutil as root

That is true, you need to add any other user that wants to access xbutil or be able to reconfigure the FPGA to the `video` group using `sudo usermod -aG video *username*`.

uday610, 2021-07-30:
closing this discussion

closed 2021-07-30 by uday610

nevilshah235, 2021-08-02:
We are working on petalinux 2020.2 based xlnx-linux kernels, we were able to generate the kernels and install them on the device. But during installation of XRT - edge zocl, we are getting the following error. 

Has anyone encountered this error before? Please can someone help to resolve the same. 

[image]