Xilinx/XRT #3252: Xilinx U250 is too chatty

holzman, 2020-04-21:
Hi,

In #2134, many of the default log levels were changed from ERR to WARN.

Consequently we have a huge amount of output in the ring buffer and on console (since the default loglevel on Linux puts DEV_WARN onto the console).  Alveo U250 is constantly sending:

```[19546.105151] hwmon hwmon2: xmc_read_from_peer: reading from peer
[19546.105170] mailbox.u mailbox.u.13631488: mailbox_request: sending request: 10 via HW
[19546.105387] mailbox.m mailbox.m.13631488: process_request: received request from peer: 10, passed on
[19546.105417] xclmgmt 0000:d8:00.0: xclmgmt_read_subdev_req: req kind 0
[19546.105506] mailbox.m mailbox.m.13631488: mailbox_post_response: posting response for: 10 via HW
[19547.098198] hwmon hwmon1: hwmon_scaling_target_temp_show: hwmon_scaling_target_temp_show: runtime clock scaling is not supported
[19547.098402] hwmon hwmon1: hwmon_scaling_target_power_show: hwmon_scaling_target_power_show: runtime clock scaling is not supported```

Can we revert this behavior? This seems like

uday610, 2021-08-17:
@chienwei-lan , can you see if these messages are still come? This issue was regarding PR #2134

chienwei-lan, 2021-08-18:
> @chienwei-lan , can you see if these messages are still come? This issue was regarding PR #2134

@uday610 , most of XRT messages are printed as KERN_DEBUG level, and PR #2134 is added another level as warning instead of treat the group of messages as error, the total amount of log printed is the same. 

If users like to make it less verbose, they can run "dmesg -n 6" to filter out KERN_DEBUG, which means only the message's priority is higher than KERN_DEBUG will be printed out.

Here is how to manipulate the debug message level
https://www.kernel.org/doc/html/latest/core-api/printk-basics.html