Xilinx/XRT #3352: Can we get support for Python2.7 dropped now?

GrueMaster, 2020-05-07:
From src/runtime_src/core/pcie/tools/xbutil/xbutil.cpp b/src/runtime_src/core/pcie/tools/xbutil/xbutil.cpp:1140
```
     // python3 is just for ppc+ubuntu setup for now. In postinst we install pyopencl with python2.7
     // so can't default to python3 if it's available. Need to revisit this when 2.7 is deprecated (Jan 1, 2020)
     #if _ARCH_PPC
         std::string cmd = "/usr/bin/python3 " + xrtTestCasePath + " -k " + xclbinPath + " -d " + std::to_string(m_idx);
     #else
         std::string cmd = "/usr/bin/python " + xrtTestCasePath + " -k " + xclbinPath + " -d " + std::to_string(m_idx);
     #endif
```
As both CentOS and Ubuntu 16.04 support Python3 natively, and most of the python libraries needed by XRT are installed via pip, it is high time to revisit this.  Currently, it is nearly impossible to install on Centos 7.xx (with current updates) as Redhat/CentOS do not have the required python libraries frozen in their repos.  Ubuntu 16.04 is ok if you apt install the python libraries (python-numpy, python-pyopencl) prior to installing xrt, but even their dependencies in pip are migrating to python3 only on a rapid basis.

GrueMaster, 2020-05-07:
My attempt at fixing this.  Patch is against 2019.2_PU1 branch.  Tested on both CentOS 7.7 and Ubuntu 16.04.6.
[XRT-python3_patch.txt](https://github.com/Xilinx/XRT/files/4594060/XRT-python3_patch.txt)

AShivangi, 2020-06-12:
We have now dropped support for python2: https://github.com/Xilinx/XRT/pull/3523

closed 2020-06-12 by AShivangi