Xilinx/XRT #3457: why I can not find sched*.bin?

rbhuang-nokia, 2020-05-26:
hi 
I want to install the XRT on centos7 and I followed the build.rst under:.../XRT-master/src/runtime_src/doc/toc/
:>src/runtime_src/tools/scripts/xrtdeps.sh
:>cd build
:>./build.sh
:>cd build/Release
:>make package
:>cd ../Debug
:>make package
when I went to 
"Before installing a locally built RPM for XRT, please copy aside /lib/firmware/xilinx/sched*.bin, which contain the ERT firmware for MicroBlaze. ......"

I cannot find "xilinx" under /lib/firmware and also the sched*.bin.Where is not corret?

Thanks

uday610, 2020-06-12:
It is because XRT never has been installed from the .deb/.rpm package.
You can download latest XRT .deb/.rpm package from Xilinx website and extract and take sched*.bin from there. 

For example, extracting .deb package
>> dpkg -x xrt_202010.2.6.655_18.04-amd64-xrt.deb mydir
Save mydir/opt/xilinx/xrt/share/fw/sched*.bin

uday610, 2020-06-12:
@stsoe , can we add this ?

rbhuang-nokia, 2020-06-12:
Thanks.I found the sched*.bin both under:/usr/lib/firmware/xilinx and /opt/xilinx/xrt/share/fw after installation the rpm pacakge according to document "ug1393-vitis-application-acceleration.pdf".

stsoe, 2021-07-15:
Resolved

closed 2021-07-15 by stsoe