Xilinx/XRT #3493: Feature Request Serial Number

rdockterjr, 2020-06-04:
This is partially a feature request and partially a question.
I'm wondering why the Alveo card Serial Number is not included in the xclDeviceInfo2 struct returned from xclGetDeviceInfo2() in src/runtime_src/core/include/xrt.h ?

This information is definitely available as shown by 
```xbutil query -d 0```
The code to do it is present in the readsensors() function of src/runtime_src/core/pcie/tools/xbutil/xbutil.h

I believe adding this functionality would be as simple as including the following inside the xclSysfsGetDeviceInfo() function of src/runtime_src/core/pcie/linux/shim.cpp
```mDev->sysfs_get("xmc", "serial_num", errmsg, info->mSerNum);```
and expanding the xclDeviceInfo2 struct to include std::string mSerNum.

This would allow users to retrieve the individual card serial numbers at runtime through libxrt_core which would be beneficial for uniquely identifying accelerator cards.

My main question is whether there is some reason not to include this information? Is this serial number available elsewhere in libxrt_core? I'm happy to file a merge request with these changes I just wanted to check with the community first.

maxzhen, 2020-06-06:
The xclDeviceInfo2 is a C data structure which is not flexible to handle the change in the future. It is not extended any more.
A JSON format is more flexible for device info. This is what you can get from the output of xbutil dump.
I think that we could probably create an API to return the same JSON formatted device info. @stsoe and @sonals , what do you think?

stsoe, 2021-07-31:
Our C++ APIs (xrt::device) support access to various device properties.   In 2021.2 `xrt::device::get_info` is extended with platform  `info::device::platform` property, which includes the serial number requested.

closed 2021-07-31 by stsoe