Xilinx/XRT #3507: XRT compilation fails due to issues with Boost 1.67 on Debian Buster

MahdiNazemi, 2020-06-09:
I am trying to install XRT after installing SDAccel 2019.1. Here is the information printed at the beginning of the installation:

-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.97
-- Looking for DRM - found at /usr 2.4.97
-- Checking for module 'OpenCL'
--   Found OpenCL, version 2.2
-- Looking for OPENCL - found at /usr 2.2 /usr/include
-- Found Git: /usr/bin/git (found version "2.20.1")
-- Looking for GIT - found at /usr/bin/git
**-- Found Boost: /usr/include (found version "1.67.0") found components: system filesystem**
-- Found Curses: /usr/lib/x86_64-linux-gnu/libcurses.so
-- XRT CL extension header files
-- include/1_2/CL/cl_ext_xilinx.h
-- include/1_2/CL/cl_ext.h
-- XRT EA eula files  /home/.XRT/src/../LICENSE
-- Platform/Linux (Debian) (Kernel 4.19.0-9-amd64)
-- Compiler: /usr/bin/c++ /usr/bin/cc
-- Found Boost: /usr/include (found version "1.67.0") found components: system filesystem program_options
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1g")
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a
-- Copying test data
-- GTest include dirs: '/usr/include'
-- Configuring CMake to run unit test after building
-- XRT xrt++ header files
-- xrt++.hpp
-- xrtexec.hpp
CMake Warning at runtime_src/ert/CMakeLists.txt:52 (message):
  ****************************************************************

  No firmware files built or copied, resulting XRT package will be missing
  ERT scheduler firmware.  Use build.sh -ertfw <dir> to specify path to a
  directory with firmware to copy during XRT build.

  ****************************************************************

When I run `build.sh -opt`, I get the following error and a few similar ones:

> /usr/bin/ld: CMakeFiles/xrt_coreutil.dir/config_reader.cpp.o: in function boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const:
> /usr/include/boost/system/error_code.hpp:703: undefined reference to boost::system::generic_category()

I would really appreciate it if you could give me a few pointers to how I could solve this issue.

keryell, 2020-07-03:
Perhaps you have different versions of Boost on your system and the link phase picked a different one from the includes you are using?
While I am using Debian/unstable on my personal laptop, I have not compiled XRT recently on it...

MahdiNazemi, 2020-07-06:
I uninstalled all different versions of boost that were installed on the server and only kept version 1.67.0.

keryell, 2020-07-06:
I gave a random try on `master` branch on `Debian/unstable` with GCC 9.3 and Boost 1.71 but the compilation fails for other reasons before:
```
/home/keryell/Xilinx/Projects/SDAccel/XRT/src/runtime_src/core/pcie/emulation/cpu_em/generic_pcie_hal2/shim.cxx: In constructor ‘xclcpuemhal2::CpuemShim::CpuemShim(unsigned int, xclDeviceInfo2&, std::__cxx11::list<xclemulation::DDRBank>&, bool, bool, FeatureRomHeader&)’:
/home/keryell/Xilinx/Projects/SDAccel/XRT/src/runtime_src/core/pcie/emulation/cpu_em/generic_pcie_hal2/shim.cxx:55:37: error: ‘int google::protobuf::MessageLite::ByteSize() const’ is deprecated: Please use ByteSizeLong() instead [-Werror=deprecated-declarations]
   55 |     ci_buf = malloc(ci_msg.ByteSize());
```

stsoe, 2020-07-11:
@akasat do you want to check the indicated compilation failure with boost 1.71?

MahdiNazemi, 2020-07-22:
> FWIW, I am able to build XRT master with boost-1.71.0 on Ubuntu 20.04 with gcc 10.1
> 
> ```
> % lsb_release -d
> Description:	Ubuntu 20.04 LTS
> % gcc --version
> gcc (Ubuntu 10-20200411-0ubuntu1) 10.0.1 20200411 (experimental)
> ```
> 
> I am using a local build of boost-1.71.0 cloned and built per src/runtime_src/tools/script/boost.sh

Installing boost-1.71.0 from `src/runtime_src/tools/script/boost.sh` certainly solves the problem.

MahdiNazemi, 2020-07-28:
However, it still fails when installing the packaged `deb` file using `gdebi` or `apt`.

stsoe, 2020-07-28:
This is a bug yes.  If you follow the build instructions in script/boost.sh, XRT will be statically linked with Boost, hence the generated package should not have a dependency on Boost, yet it does.   script/boost.sh is experimental to facilitate static linking of Boost, the DEB/RPM package scripts were not updated accordingly.  I will look at this.

What I don't understand is your problems building XRT to begin with?  If you are on one of the supported OSes, you must run scripts/xrtdeps.sh to install the build time dependencies.    From your original error about boost::system::generic_category it appears your server is a custom setup and that some Boost headers are missing or non standard.   I recently installed Ubuntu20.04 and were able to build XRT and install the resulting DEB with apt install no problem.

MahdiNazemi, 2020-07-29:
> What I don't understand is your problems building XRT to begin with? If you are on one of the supported OSes, you must run scripts/xrtdeps.sh to install the build time dependencies. From your [...]

I am running the latest version of Debian and its corresponding Boost library (i.e. 1.67.0).

The required dependencies are installed when I run `xrtdeps.sh` and I haven't noticed any errors/warnings during the installation of dependencies.

stsoe, 2020-07-29:
XRT officially supports only the systems listed here: https://xilinx.github.io/XRT/master/html/system_requirements.html 
While we appreciate the notification of problems building XRT on different distributions, it's difficult for us to keep up with all the distributions.  

Wanting to duplicate your problem, I created a docker container from debian:latest, I used XRT:master, ran xrtdeps.sh, and then build.sh -opt,  Everything built and I was able to also install the .deb.

```
soeren@e90a7697ae92:~/git/stsoe/XRT.u20$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
soeren@e90a7697ae92:~/git/stsoe/XRT.u20$ dpkg -l xrt
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=======================================================
ii  xrt            2.8.0        amd64        Xilinx RunTime stack for use with Xilinx FPGA platforms
```

keryell, 2020-07-29:
I appreciate the effort!

MahdiNazemi, 2020-07-29:
> XRT officially supports only the systems listed here: https://xilinx.github.io/XRT/master/html/system_requirements.html
> While we appreciate the notification of problems building XRT on different [...]

Thank you very much for taking the time to investigate this issue. I will try to find the reason why it doesn't work on my machine. Meanwhile, would you please share the Boost version of your sicker?

stsoe, 2020-07-29:
```
soeren@e90a7697ae92:~/git/stsoe/XRT.u20$ dpkg -l |grep -i boost
ii  libboost-dev:amd64                     1.67.0.1                    amd64        Boost C++ Libraries development files (default version)
ii  libboost-filesystem-dev:amd64          1.67.0.1                    amd64        filesystem operations (portable paths, iteration over directories, etc) in C++ (default version)
ii  libboost-filesystem1.67-dev:amd64      1.67.0-13+deb10u1           amd64        filesystem operations (portable paths, iteration over directories, etc) in C++
ii  libboost-filesystem1.67.0:amd64        1.67.0-13+deb10u1           amd64        filesystem operations (portable paths, iteration over directories, etc) in C++
ii  libboost-program-options-dev:amd64     1.67.0.1                    amd64        program options library for C++ (default version)
ii  libboost-program-options1.67-dev:amd64 1.67.0-13+deb10u1           amd64        program options library for C++
ii  libboost-program-options1.67.0:amd64   1.67.0-13+deb10u1           amd64        program options library for C++
ii  libboost-system1.67-dev:amd64          1.67.0-13+deb10u1           amd64        Operating system (e.g. diagnostics support) library
ii  libboost-system1.67.0:amd64            1.67.0-13+deb10u1           amd64        Operating system (e.g. diagnostics support) library
ii  libboost1.67-dev:amd64                 1.67.0-13+deb10u1           amd64        Boost C++ Libraries development files
```

keryell, 2021-03-10:
@MahdiNazemi Is this issue still a problem?

MahdiNazemi, 2021-03-10:
@keryell I spent hours trying to fix the issue but was unsuccessful. Since @stsoe did not face any issues while installing the software on a clean OS, I suspect there is a conflict with one of the packages on my machine. Please feel free to close the issue when you see fit.

keryell, 2021-05-22:
@MahdiNazemi I guess you can close this by yourself.

closed 2021-06-30 by MahdiNazemi