Xilinx/XRT #3541: error: ‘MAP_HUGE_SHIFT’ was not declared in this scope

malith199122, 2020-06-12:
When I tried to build the XRT with the build script (build.sh) I got the following error
XRT/src/runtime_src/core/pcie/linux/shim.cpp: In member function ‘int xocl::shim::cmaEnable(bool, uint64_t)’:
XRT/src/runtime_src/core/pcie/linux/shim.cpp:1215:140: error: ‘MAP_HUGE_SHIFT’ was not declared in this scope
void *addr_local = mmap(0x0, page_sz, PROT_READ | PROT_WRITE, MAP_PRIVATE | MAP_ANONYMOUS | MAP_HUGETLB | hugepage_flag << MAP_HUGE_SHIFT, 0, 0);

Thanks in advance

chienwei-lan, 2020-06-16:
Hello @malith199122, could you please share the OS/kernel version you are using?

run uname -a & uname -r in terminal(assume you are using Linux)

There is a quick hack to pass the check:

Add this line at XRT/src/runtime_src/core/pcie/linux/shim.cpp:83(or any where)

**#define MAP_HUGE_SHIFT 26** 

It should fix the compile issue

sonals, 2021-11-05:
@malith199122 is this issue resolved with standard Linux install? Do you still the hack?