Xilinx/XRT #3626: Two tests failing

boxerab, 2020-06-25:
I already have XRT installed from binary, btw.

```
Test project /home/aaron/src/XRT/build/Debug
    Start 3: xbutil2
    Start 4: xbmgmt2
    Start 2: xbutil
    Start 1: xbmgmt
1/4 Test #3: xbutil2 ..........................***Failed    0.00 sec
/home/aaron/src/XRT/build/Debug/runtime_src/core/tools/xbutil2/xbutil2: symbol lookup error: /home/aaron/src/XRT/build/Debug/runtime_src/core/tools/xbutil2/xbutil2: undefined symbol: xclGetDebugIpLayout

2/4 Test #4: xbmgmt2 ..........................***Failed    0.00 sec
/home/aaron/src/XRT/build/Debug/runtime_src/core/tools/xbmgmt2/xbmgmt2: symbol lookup error: /home/aaron/src/XRT/build/Debug/runtime_src/core/tools/xbmgmt2/xbmgmt2: undefined symbol: xclGetDebugIpLayout

3/4 Test #1: xbmgmt ...........................   Passed    0.01 sec
4/4 Test #2: xbutil ...........................   Passed    0.01 sec

50% tests passed, 2 tests failed out of 4

Total Test time (real) =   0.01 sec

The following tests FAILED:
	  3 - xbutil2 (Failed)
	  4 - xbmgmt2 (Failed)
```

boxerab, 2020-06-29:
Fixed now.

closed 2020-06-29 by boxerab

Sriengchhun, 2021-03-02:
How can you fix this problem?