Xilinx/XRT #3650: XRT 2019.2: WARNING: No recipes available for: petalinux-image.bbappend

lucasbrasilino, 2020-06-30:
Hello,

I'm following instructions below to build XRT 2019.2 for a ZCU102. I'm working on `2019.2` branch, using Petalinux 2019.2 and Vitis 2019.2, on a Ubuntu 18.04 LTS host.

https://xilinx.github.io/XRT/2019.2/html/yocto.html#create-petalinux-project-with-xrt-recipes

However, in my project there is no `project-spec/meta-user/recipes-core/images/petalinux-image.bbappend` file. Even even though I've created it and added the instructed content.

When running `petalinux-build` (after running `petalinux-config` for kernel and rootfs), I get the message:

```bash
INFO: bitbake petalinux-user-image
Loading cache: 100% |##########################################################################################################################################################| Time: 0:00:00
Loaded 3981 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################| Time: 0:00:03
Parsing of 2897 .bb files complete (2891 cached, 6 parsed). 3987 targets, 155 skipped, 0 masked, 0 errors.
WARNING: No recipes available for:
  <PATH_TO_MY_PROJECT>/project-spec/meta-user/recipes-core/images/petalinux-image.bbappend
```

and no `zocl` neither XRT binaries are created:

```bash
$ tar tzvf rootfs.tar.gz | egrep "zocl|xrt|opencl"
```

Any help?
Thanks!

larry9523, 2020-06-30:
Can you try the name petalinux-image-full.bbappend instead of petalinux-image.bbappend

lucasbrasilino, 2020-06-30:
Tried the suggestion. What I did:

* Cleaned project with `petalinux-build -x mrproper -f`.
* Reconfigured again with `petalinux-config --get-hw-description=<path to HDF dir>
* added the `project-spec/meta-user/recipes-core/images/petalinux-image-full.bbappend` file
* Ran `petalinux-config -c kernel`
* Ran `petalinux-config -c rootfs`
* Ran `petalinux-build`

No warning was issued. But XRT binaries/module weren't built:

(running `find` from project's root directory)

```bash
$ tar tzvf rootfs.tar.gz | egrep "zocl|xrt|opencl"
$ find . -name '*zocl*'
./project-spec/meta-user/recipes-xrt/zocl
./project-spec/meta-user/recipes-xrt/zocl/zocl_git.bb
./components/plnx_workspace/sources/linux-xlnx/drivers/gpu/drm/zocl
./components/plnx_workspace/sources/linux-xlnx/drivers/gpu/drm/zocl/zocl_drv.h
./components/plnx_workspace/sources/linux-xlnx/drivers/gpu/drm/zocl/zocl_bo.c
./components/plnx_workspace/sources/linux-xlnx/drivers/gpu/drm/zocl/zocl_drv.c
./components/plnx_workspace/sources/linux-xlnx/include/uapi/linux/zocl_ioctl.h
./components/plnx_workspace/sources/linux-xlnx/Documentation/devicetree/bindings/drm/zocl
./components/plnx_workspace/sources/linux-xlnx/Documentation/devicetree/bindings/drm/zocl/zocl_drm.txt
$ find . -name '*xrt*'
./project-spec/meta-user/recipes-xrt
./project-spec/meta-user/recipes-xrt/xrt
./project-spec/meta-user/recipes-xrt/xrt/xrt_git.bb
```

Any further suggestions ?

larry9523, 2020-06-30:
1) Do you have project-spec/meta-plnx-generated/recipes-core/images/petalinux-user-image.bb ? Is there zocl and xrt in your IMAGE_INSTALL=" " variable?

2) Or, did you add 
"
IMAGE_INSTALL_append = " xrt"
IMAGE_INSTALL_append = " zocl"
"
in your project-spec/meta-user/recipes-core/images/petalinux-image-full.bbappend

lucasbrasilino, 2020-06-30:
1. Yes, I do have a `project-spec/meta-plnx-generated/recipes-core/images/petalinux-user-image.bb` but there's no `zoctl` and `xrt` on it. Here goes its content:

```bash
$ cat project-spec/meta-plnx-generated/recipes-core/images/petalinux-user-image.bb
DESCRIPTION = "PETALINUX image definition for Xilinx boards"
LICENSE = "MIT"

require recipes-core/images/petalinux-image-common.inc 

inherit extrausers 
COMMON_FEATURES = "\
		ssh-server-dropbear \
		hwcodecs \
		"
IMAGE_LINGUAS = " "

IMAGE_INSTALL = "\
		kernel-modules \
		haveged \
		mtd-utils \
		canutils \
		openssh-sftp-server \
		pciutils \
		run-postinsts \
		udev-extraconf \
		packagegroup-core-boot \
		packagegroup-core-ssh-dropbear \
		tcf-agent \
		watchdog-init \
		bridge-utils \
		hellopm \
		"
EXTRA_USERS_PARAMS ?= "usermod -P root root;"
```

2. I added those `IMAGE_INSTALL_append` to the `petalinux-image-full.bbappend` as described in [create-petalinux-project-with-xrt-recipes](https://xilinx.github.io/XRT/2019.2/html/yocto.html#create-petalinux-project-with-xrt-recipes) doc.

[Please find the `petalinux-image-full.bbappend` zipfile using this link](https://github.com/Xilinx/XRT/files/4854566/petalinux-image-full.bbappend.zip). You can download, unzip and check its content.

Thanks!

larry9523, 2020-06-30:
Assuming we have a valid xrt_git.bb file, what does it say if you run
$petalinux-build -c xrt

lucasbrasilino, 2020-06-30:
It ran ok and built libraries:

```bash
$ petalinux-build -c xrt
[INFO] building xrt
[INFO] sourcing bitbake
[INFO] generating user layers
[INFO] generating workspace directory
INFO: bitbake xrt
Loading cache: 100% |##########################################################################################################################################################| Time: 0:00:00
Loaded 3981 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################| Time: 0:00:03
Parsing of 2897 .bb files complete (2892 cached, 5 parsed). 3987 targets, 155 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |#####################################################################################################################################################| Time: 0:00:01
Checking sstate mirror object availability: 100% |#############################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 148 Found 47 Missed 202 Current 554 (31% match, 85% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 2203 tasks of which 2147 didn't need to be rerun and all succeeded.
INFO: Copying Images from deploy to images
NOTE: Failed to copy built images to tftp dir: /tftpboot 
[INFO] successfully built xrt
```

```bash
$ find . -name '*xrt*'
[... lots of stuff ...]
./build/tmp/sysroots-components/aarch64/xrt
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt/xrt_mem.h
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt/experimental/xrt-next.h
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt/experimental/xrt++.hpp
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt/experimental/xrtexec.hpp
./build/tmp/sysroots-components/aarch64/xrt/usr/include/xrt/xrt.h
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_swemu.so.2
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_core.so
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_coreutil.so.2
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_core.so.2
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_hwemu.so.2
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt++.so.2.3.0
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_core.so.2.3.0
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt++.so.2
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_coreutil.so
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_hwemu.so.2.3.0
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_hwemu.so
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt++.so
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_swemu.so
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_swemu.so.2.3.0
./build/tmp/sysroots-components/aarch64/xrt/usr/lib/libxrt_coreutil.so.2.3.0
./build/tmp/sysroots-components/aarch64/xrt/sysroot-providers/xrt
[... lots of stuff ...]
```

larry9523, 2020-06-30:
hmm, seems you are all set. :) I have no idea why full "petalinux-build" won't give you xrt but "petalinux-build -c xrt" does.
@mamin506 @chvamshi-xilinx, any clue?

one more thing you may want to try is delete the entire "./build" folder and rerun "petalinux-build"

larry9523, 2020-06-30:
oh, one more thing you can try is check your project-spec/configs/rootfs_config add following if they are not there
#
# xrt
#
CONFIG_xrt=y
CONFIG_xrt-dev=y
# CONFIG_xrt-dbg is not set

#
# zocl
#
CONFIG_zocl=y
CONFIG_zocl-dev=y
# CONFIG_zocl-dbg is not set

mamin506, 2020-06-30:
Hi @lucasbrasilino , please confirm if you missed below step in [create petalinux with xrt recipes](https://xilinx.github.io/XRT/2019.2/html/yocto.html#create-petalinux-project-with-xrt-recipes)

```
# Configure rootfs, enable below components:
#   menu -> "user packages" -> xrt
#   menu -> "user packages" -> xrt-dev
#   menu -> "user packages" -> zocl
#   menu -> "user packages" -> opencl-headers-dev
#   menu -> "user packages" -> opencl-clhpp-dev
$ petalinux-config -c rootfs
```

This is equivalence to what Larry suggested in rootfs_config. This step is to tell petalinux to build and install packages into rootfs.

lucasbrasilino, 2020-07-01:
Hi @mamin506 , 
Those options does **not** appear in my rootfs configuration menu, as you can see _*below*_. However, in the `project-spec/configs/rootfs_config` their options show up:

```
#
# xrt
#
# CONFIG_xrt is not set
# CONFIG_xrt-dev is not set
# CONFIG_xrt-dbg is not set

#
# zocl
#
# CONFIG_zocl is not set
# CONFIG_zocl-dev is not set
# CONFIG_zocl-dbg is not set
```

**What can be happening for them to not show in menu?**

@larry9523 I will try setting them manually in `rootfs_config` file. Thanks.

[image]

mamin506, 2020-07-01:
If you didn't see them on the menu, it means you needs to do below step in petalinux 2019.2
Open `project-spec/meta-user/conf/user-rootfsconfig`
Add below lines
`
CONFIG_xrt
CONFIG_xrt-dev
CONFIG_zocl
CONFIG_opencl-clhpp-dev
CONFIG_opencl-headers-dev
`
In 2019.1 petalinux, it is the change Larry mentioned in `project-spec/meta-user/recipes-core/images/petalinux-image-full.bbappend`.
In before 2019.1 petalinux, it is in `project-spec/meta-user/recipes-core/images/petalinux-image.bbappend`

lucasbrasilino, 2020-07-01:
With those modifications I am able to generate the rootfs with XRT. Thank all of you for the help.

closed 2020-07-01 by lucasbrasilino