Xilinx/XRT #3654: libstdc++ version conflict with Vitis 2020.1 and XRT master

boxerab, 2020-07-01:
Running this command:

`v++ -t hw --platform xilinx_u250_xdma_201830_2 --link Kernel.xo -o'Kernel.xclbin'`

I get this error at the very end:

```
/opt/xilinx/xrt/bin/unwrapped/xclbinutil: /tools/Xilinx/Vitis/2020.1/lib/lnx64.o/Ubuntu/18/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /opt/xilinx/xrt/bin/unwrapped/xclbinutil)
INFO: [v++ 60-1442] [13:55:40] Run run_link: Step xclbinutil: Failed
```

My output from `strings`:

```
$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6  | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

boxerab, 2020-07-01:
The problem was that Vitis 2020.1 installs `libstdc++.s0.6` in two places:
```
./lib/lnx64.o/Ubuntu/
./lib/lnx64.o/Ubuntu/18/
```

As I am using GCC 10, and I built XRT locally, `xclbinutil` is expecting a newer version of `libstdc++.so.6`, but apparently the Vitis
.so files are being loaded instead of the system .so files.

My solution was to delete the .so files in the two directories above.

closed 2020-07-01 by boxerab

keryell, 2020-07-02:
It would be interesting to keep track of the full recipe to use a Xilinx FPGA in a modern environment.