Xilinx/XRT #3663: Error when compiling with hardware emulation

boxerab, 2020-07-03:
`v++ -g --profile_kernel data:all:all:all -t hw_emu --platform xilinx_u250_xdma_201830_2 -c -k Kernel -I'./' ./Kernel.cl -o Kernel250HWEMU.xo`

```
ERROR: [VPL 60-773] In 'vpl/runme.log', caught Tcl error:  ./elaborate.sh: line 20: LIBRARY_PATH: unbound variable
ERROR: [VPL 60-2373] In '_x/link/vivado/vpl/runme.log', caught error: ERROR: caught error: ./elaborate.sh: line 20: LIBRARY_PATH: unbound variable
ERROR: [VPL 60-2373] In '_x/link/vivado/vpl/vivado.log', caught error: ERROR: caught error: ./elaborate.sh: line 20: LIBRARY_PATH: unbound variable
```

boxerab, 2020-07-03:
Solution:

`export LIBRARY_PATH=$LIBRARY_PATH:/usr/lib/x86_64-linux-gnu`

keryell, 2020-07-03:
Is it something that should be documented somewhere?

boxerab, 2020-07-04:
> Is it something that should be documented somewhere?

Yes, I think so. This could go in `/opt/xilinx/xrt/setup.sh`