Xilinx/XRT #3669: xclbin from software/hardware emu mode cannot be loaded by clBuildProgram

boxerab, 2020-07-06:
I am only able to load the kernel in from hardware link, not from emulation.

stsoe, 2020-07-11:
@boxerab Please use clCreateProgramWithBinary to load xclbin.   It is necessary to set XCL_EMULATION_MODE to either sw_emu or hw_emu if the xclbin was created for either.

boxerab, 2020-07-11:
Thanks. Actually, my mistake, I did use `clCreateProgramWithBinary`. So, I need to set `XCL_EMULATION_MODE` when calling
`clCreateProgramWithBinary` ?

stsoe, 2020-07-11:
Yes,  run your application with XCL_EMULATION_MODE set per how the xclbin was created.
FYI, we do have a change request for determining the flow based on the xclbin metadata, but today you have to set the env var.

boxerab, 2020-07-11:
Great, thanks. Yes, it would be handy to deduce emulation from the binary.

closed 2020-07-11 by boxerab