Xilinx/XRT #3756: XRT does not support Boost 1.73.0

paroque28, 2020-07-28:
XRT fails in compilation:
```
fatal error: boost/  detail/endian.hpp: No such file or directory | #include <boost/detail/endian.hpp>
```
In 1.73.0 some macros and header files were deprecated.
For example:
`BOOST_BYTE_ORDER`
:
```
The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
```

stsoe, 2020-07-29:
Correct we don't support boost 1.73.0.  A pull request would be greatly appreciated :-)

keryell, 2020-07-29:
Actually I have the same issue with `Debian/unstable` which comes with 1.71.

stsoe, 2020-07-29:
I am positive we can build with 1.71, because that is the version used on Ubuntu 20.04.  You will get a deprecation warning but not a hard error.   Regardless this of course needs to be fixed.  The problem is we need to support boost all the way back to 1.53.0 (CentOS7.x), I may just disable the compilation check.

stsoe, 2020-11-11:
References to endian.hpp were removed

closed 2020-11-11 by stsoe