Xilinx/XRT #3767: XMA initialization fails with exception with software or hardware emulation

gbonneau-hardent, 2020-07-29:
On Linux the XMA initialization will fail with an exception when software or hardware emulation is requested:

Reproduce steps: Use the copy_encoder example of repository https://github.com/Xilinx/xma-samples . When the datamover application call xma_initialize() in the application with an xrt.ini configured for software emulation and you got the message:

terminate called after throwing an instance of 'std::runtime_error'
  what():  singleton ctor error
Signal: SIGABRT (Aborted)

This is caused by the linking of the host application with both xma2api and xrt_core libraries. the xma2api needs the xrt_core to read the configuration file. For example in xma_utils.cpp this call is made: xrt_core::config::get_sw_em_driver().  This will create a singleton that is bound with xrt_core. Yet as soon as the hardware/software emulation shared library has dynamically loaded the creation of the singleton for the software/hardware emulation shared library will throw an exception. This singleton was already created by the xrt_core shared library.

stsoe, 2020-07-29:
Host application for XMA and OpenCL should not link with xrt_core library.  XMA and OpenCL dlopens the proper xrt_core library depending on the value of XCL_EMULATION mode.    It's a confusing that xrt_core::config appears to be in the xrt_core library, in fact it is in xrt_coreutil so no need to link with xrt_core.

gbonneau-hardent, 2020-07-29:
I tried with the xilinx xma-samples. Dependencies with xrt_core remains and yield undefined reference. Using xrt_coreutil library only yield those undefined:

/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclExecWait'                                                                                                                                                                                                                                                                                                                                                                      
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclLogMsg'                                                                                                                                                                                                                                                                                                                                                                        
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclAllocBO'                                                                                                                                                                                                                                                                                                                                                                       
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `pcidev::pci_device::sysfs_get(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclOpenContext'                                                                                                                                                                                                                                                                                                                                                                   
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclFreeBO'                                                                                                                                                                                                                                                                                                                                                                        
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclLoadXclBin'                                                                                                                                                                                                                                                                                                                                                                    
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `pcidev::get_dev(unsigned int, bool)'                                                                                                                                                                                                                                                                                                                                              
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `pcidev::get_dev_total(bool)'                                                                                                                                                                                                                                                                                                                                                      
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclProbe'                                                                                                                                                                                                                                                                                                                                                                         
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclOpen'                                                                                                                                                                                                                                                                                                                                                                          
/git/XRT/build/Debug/opt/xilinx/xrt/lib/libxma2api.so: undefined reference to `xclMapBO'

stsoe, 2020-07-29:
Oh, yes that is no good.  Thanks for pointing this out.

gbonneau-hardent, 2020-08-05:
After pondering about the undefined reference I remembered that under Linux (not sure for Windows) that you can delay the binding of the undefined reference until the shared library is loaded with **_dlopen_** . However you need to tell the linker to don't care about the undefined symbols with adding the linker option: -Wl,--allow-shlib-undefined . I am then able to have the software emulation library loaded. So far so good. I was very close to have a successful running session in software emulation mode. 

However the **_xma_initialize_** request the hardware configuration with a call to xma_hw_configure(). Then the configure function calls the hardware HAL **_hal_configure_**() to retrieve information from the xclbin file which was created with a TARGET of **_sw_emu_** . The configuration routine then proceeds to read metadata from the header section of the xclbin file. For example MEM_TOPOLOGY, CONNECTIVITY, IP_LAYOUT which are not available in software emulation. At this point, an error is returned and the application aborts. Note that I am using the git commit with tag 2019.2.0.3 which is the one used by Amazon for AWS F1.

I have opened issue #3775 for XMA not being able to instantiate a session for software emulation. I need to understand if minor modifications to the code are enough for successful software emulation at this point. For an accelerator board like Alveo, the hardware compilation of the code take almost 3 hours. This is insane and obviously software emulation is the way to go for the design before trying on a real hardware platform. Could you please comment on how to get the XMA software emulation working.? Is the XMA software emulation has ever been tried?

stsoe, 2020-08-05:
I don't think XMA should have dependencies on libxrt_core directly, that's the root of the problem complicating emulation.   I am not keen on allowing shared objects with undefined symbols, but as you point out here may be the quickest work-around here, e.g. just dlopen the core library needed based on whatever flow you want to run.

We are in the process of adding native APIs to XRT that do not depend on xrt_core, yet provide the necessary functionality.  XMA should be switched to use the native APIs and remove all dependencies on xrt_core.

sarab96, 2020-08-05:
XMA support hardware emulation. For software emulation use Xilinx's OpenCL library

closed 2020-08-05 by sarab96