Xilinx/XRT #3770: Use of "this pointer" in base constructor of xrt_core::system will fail calls of virtual derived function.

gbonneau-hardent, 2020-07-29:
In system.cpp the this pointer is used in the base class constructor of xrt_core::system. Any derived class from base class system will fail when virtual method of derived class will be called. 

`system::
system()
{
  if (singleton)
    throw std::runtime_error("singleton ctor error");
  singleton = this;
}
`

From C++ FAQ 10.7:

**_> Here is something that always works: the {body} of a constructor (or a function called from the constructor) can reliably access the data members declared in a base class and/or the data members declared in the constructor's own class. This is because all those data members are guaranteed to have been fully constructed by the time the constructor's {body} starts executing.
> 
> Here is something that never works: the {body} of a constructor (or a function called from the constructor) cannot get down to a derived class by calling a virtual member function that is overridden in the derived class. If your goal was to get to the overridden function in the derived class, you won't get what you want. Note that you won't get to the override in the derived class independent of how you call the virtual member function: explicitly using the this pointer (e.g., this->method()), implicitly using the this pointer (e.g., method()), or even calling some other function that calls the virtual member function on your this object. The bottom line is this: even if the caller is constructing an object of a derived class, during the constructor of the base class, your object is not yet of that derived class. You have been warned._**

stsoe, 2020-07-29:
Hi @gbonneau-hardent The FAQ is correct, but it talks about using virtual functions **within** the the base class constructor.  The base class constructor is called before the derived class is constructed, so it can't call virtual methods.  The system() constructor is not using the derived class.  Later use of singleton will work as expected because the derived class has been constructed.

gbonneau-hardent, 2020-07-29:
Yes. Agree. Let close it.

closed 2020-07-29 by gbonneau-hardent