Xilinx/XRT #3775: xma doesn't support software and hardware emulation

gbonneau-hardent, 2020-07-29:
While XMA is initialized the hardware configuration is probed with a call to xma_hw_probe(). However, a Hardware Abstraction Layer (HAL) is missing for software of hardware emulation which prevents the development and debugging of applications using emulation.

stsoe, 2020-08-05:
As regards sw_emu, this is almost impossible to get to work without a lot of conditional code to work-around missing IP_LAYOUT, CONNECTIVITY, and MEM_TOPOLOGY.  We need to include these sections in the xclbin, but that will happen earliest 2021.1.   In reference to my comment in #3767, the native APIs that will be added in 2020.1 will also not support sw_emu for exactly this reason (missing sections in the xclbin).

sarab96, 2020-08-05:
XMA support hardware emulation. For software emulation use Xilinx's OpenCL library

closed 2020-08-05 by sarab96