Xilinx/XRT #3818: XRT cannot find newly-installed QDMA platform

Hecmay, 2020-08-07:
Hi, I have a U250 board with QDMA shell installed. 

When running the compiled bitstream, the XRT complains that it cannot find the correct device. Here is the error message:

```shell
Found Platform
Platform Name: Xilinx
INFO: Reading build_dir.hw.xilinx_u250_qdma_201920_1/krnl_incr.xclbin
Loading: 'build_dir.hw.xilinx_u250_qdma_201920_1/krnl_incr.xclbin'
Trying to program device[0]: xilinx_u250_xdma_201830_2
XRT build version: 2.3.1301
Build hash: 192e706aea53163a04c574f9b3fe9ed76b6ca471
Build date: 2019-10-25 03:04:42
Git branch: 2019.2
PID: 23884
UID: 1225572
[Thu Aug  6 20:05:09 2020]
[XRT] ERROR: Xclbin does not match Shell on card.
[XRT] ERROR: Use 'xbmgmt flash' to update Shell.
[XRT] ERROR: See dmesg log for details. err=-95
[XRT] ERROR: Failed to load xclbin.
Failed to program device[0] with xclbin file!
Trying to program device[1]: xilinx_u280_xdma_201920_1
[XRT] ERROR: Xclbin does not match Shell on card.
[XRT] ERROR: Use 'xbmgmt flash' to update Shell.
[XRT] ERROR: See dmesg log for details. err=-95
[XRT] ERROR: Failed to load xclbin.
Failed to program device[1] with xclbin file!
Failed to program any device found, exit!
```

I have flashed the card after installing the QDMA shells. How should I reconfigure XRT to let it recognize the new platform?

mariodruiz, 2020-08-17:
Hi @Hecmay,

The card seems not to be flashed with the qdma shell see this line `Trying to program device[0]: xilinx_u250_xdma_201830_2`
What is the output of `sudo xbmgmt flash --scan`?

I think this is not an issue with XRT, but that the Alveo is not flashed properly. Did you cold-reboot after flashing qdma shell?

uday610, 2021-07-31:
closed as not XRT issue

closed 2021-07-31 by uday610