Xilinx/XRT #3894: Concurrent kernels not being generated

boxerab, 2020-08-19:
Following the documentation [here]https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/buildingdevicebinary.html#yzb1524519238289, I have add a cfg file with line

`nk=FOO:3`

but, when I call `clCreateKernel` with `FOO_1`, I get an error. `FOO` does work.

What am I missing ?
Thanks!

boxerab, 2020-08-20:
It looks like:

`FOO:{FOO_1}` will work. The documents should probably be updated to reflect this.

uday610, 2020-09-15:
@boxerab , that syntax is documented in host coding section (your link is pointing to the kernel compilation section).

https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/devhostapp.html#kol1557424570275__section_kxw_rvc_ljb

closed 2020-10-13 by boxerab