Xilinx/XRT #3953: Thread-safety issue with enqueueMapBuffer() and parallel buffer allocation on Alveo boards

zohourih, 2020-08-28:
Trying to debug a "Bus error (core dumped)" crash on our Alveo U50 board, I realized that it seems there is some thread-safety issue with using the `enqueueMapBuffer()` API followed by OpenMP-parallelized buffer allocation using the host pointer, which is easily reproducible. Disabling OpenMP allows the code to run without any issues. The same code with OpenMP parallelization runs fine on MPSoC boards (which doesn't involve host to device buffer copy due to the memory being shared between the FPGA and the ARM CPU). It seems XRT fails to correctly handle the different OpenMP threads when explicitly copying data from the host buffer to the device across the PCI-E bus on Alveo boards.

To reproduce the issue, simply add `#pragma omp parallel for` to line 131 of the Vitis Accel `vadd` example [here](https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/cpp_kernels/simple_vadd/src/vadd.cpp) (or any other example that uses this type of buffer allocation/movement) and add `-fopenmp` to CXXFLAGS in the `Makefile` and compile it and run. This will reliably result in a "Bus error (core dumped)" crash. This seems like a serious thread-safety issue in XRT. based on the OpenCL specification, every OpenCL function except `clSetKernelArg()` are supposed to be thread-safe.

Tested with Ubuntu 18.04, Alveo U50, XRT 2.7.766 (taken from [here](https://www.xilinx.com/products/boards-and-kits/alveo/u50.html#xrt1804ubuntu)), both Vitis 2019.2 and 2020.1.

stsoe, 2020-10-23:
Thank you for this bug report.    The issue is not OpenCL specific.  It is fixed by #4297 .

closed 2020-10-23 by stsoe

zohourih, 2020-10-23:
Thank you for the fix.