Xilinx/XRT #4003: Error when installing U50 deployment package

boxerab, 2020-09-04:
This is create_xsabin.sh running from /usr/lib/firmware/xilinx/f465b0a3ae8c64f619bc150384ace69b on Fri 04 Sep 2020 04:03:45 PM EDT
Run for install of partition xilinx-u50-gen3x16-xdma-blp version 1 release 2784799
Metadata file is partition_metadata.json
User install path is /opt/xilinx/firmware/u50/gen3x16-xdma/blp
Required ert firmware install directory not found at /opt/xilinx/xrt/share/fw. Unable to build xsabin files

zohourih, 2020-10-06:
If you are using a supported version of CentOS or Ubuntu, you can just use the pre-compiled XRT package from here:
https://www.xilinx.com/products/boards-and-kits/alveo/u50.html#gettingStarted

If you want to build XRT from source, you should follow the build instructions posted here:
https://xilinx.github.io/XRT/master/html/build.html
Specifically, you need Vitis installed and `XILINX_VITIS` set to the correct path for the build process to be able to build the firmware.

boxerab, 2020-10-07:
Thanks @zohourih . I am using the latest XRT release binary along with supported version of Ubuntu.
But, I still get this error.

zohourih, 2020-10-07:
Have you installed all the three packages in the "Deployment Target Platform" package? The firmware is installed by one of those.

boxerab, 2020-10-07:
Thanks, this issue is now resolved. Card now validates and is recognized by `clinfo`.

closed 2020-10-07 by boxerab