Xilinx/XRT #4029: xbutil query segmentation fault

acdy147, 2020-09-10:
Hello,

$ /opt/xilinx/xrt/bin/xbutil query terminates with SIGSEGV.

Ubuntu 18.04.5 LTS 
Kernel 4.15.0-115-generic
XRT 2.3.1301

It works fine on 4.15.0-112-generic. 

Please advise. 

(gdb) bt
#0  0x000055cda17d5ab3 in pcidev::pci_device::get_sysfs_path(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#1  0x000055cda17d6b3e in pcidev::pci_device::sysfs_open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool, bool) ()
#2  0x000055cda17d741a in pcidev::pci_device::sysfs_get(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&) ()
#3  0x000055cda17d7724 in pcidev::pci_device::sysfs_get(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::vector<unsigned long, std::allocator<unsigned long> >&) ()
#4  0x000055cda17dcfc6 in xocl::shim::xclSysfsGetDeviceInfo(xclDeviceInfo2*) ()
#5  0x000055cda17df8e4 in xocl::shim::xclGetDeviceInfo2(xclDeviceInfo2*) ()
#6  0x000055cda178ab4d in xcldev::device::device(unsigned int, char const*) ()
#7  0x000055cda1775463 in main ()

uday610, 2020-09-15:
@acdy147, this is very old release, I recommend you to try the latest 2020.1_PU1 release (downloadable from xilinx.com) or even 2020.1 release.  There maybe something else going on, so it will be good if we can check with latest release

stsoe, 2021-07-15:
2.3.x no longer supported, please use later release.

closed 2021-07-15 by stsoe