Xilinx/XRT #4036: Broken links in the .rst documentation

keryell, 2020-09-11:
For example on https://github.com/Xilinx/XRT/blob/master/src/runtime_src/doc/toc/system_requirements.rst the links are broken.

uday610, 2020-10-11:
when linking one .rst file from another .rst file, I have seen the link does not work. After creating the HTML files those work. You can check the HTML files https://xilinx.github.io/XRT/index.html there link should be working

keryell, 2020-10-11:
Could you make them to work for both, like in https://github.com/triSYCL/triSYCL/blob/master/doc/testing.rst ?

keryell, 2020-10-13:
This is probably a limitation of the GitHub ReST renderer with the `:ref:` syntax

uday610, 2021-09-15:
I unable to fix this, hence closing. When we have a techdoc expert we can address this issue.

closed 2021-09-15 by uday610