Xilinx/XRT #4131: XRT Build Failure Caused By Deprecated Protobuf API Calls

jo7ueb, 2020-09-30:
I encountered following error when running `build.sh`.
Deprecated Protobuf API is called many times.

```
[ 19%] Building CXX object xma/src/xmaapi/CMakeFiles/xma2api.dir/xmalogger.cpp.o
In file included from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:28,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx: In member function ‘void xclhwemhal2::HwEmShim::fetchAndPrintMessages()’:
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:35:38: error: ‘int google::protobuf::MessageLite::ByteSize() const’ is deprecated: Please use ByteSizeLong() instead [-Werror=deprecated-declarations]
   35 |      unsigned c_len = c_msg.ByteSize(); \
      |                                      ^
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:529:3: note: in expansion of macro ‘SERIALIZE_AND_SEND_MSG’
  529 |   SERIALIZE_AND_SEND_MSG(func_name) \
      |   ^~~~~~~~~~~~~~~~~~~~~~
/home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:186:4: note: in expansion of macro ‘xclGetDebugMessages_RPC_CALL’
  186 |    xclGetDebugMessages_RPC_CALL(xclGetDebugMessages,ack,force,displayMsgs,logMsgs,stopMsgs);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/google/protobuf/generated_enum_util.h:36,
                 from /usr/local/include/google/protobuf/map.h:55,
                 from /usr/local/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/yasu/xilinx/XRT/build/Debug/runtime_src/core/pcie/emulation/hw_em/../common_em/rpc_messages.pb.h:26,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:25,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/usr/local/include/google/protobuf/message_lite.h:420:7: note: declared here
  420 |   int ByteSize() const { return internal::ToIntSize(ByteSizeLong()); }
      |       ^~~~~~~~
In file included from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:28,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:42:39: error: ‘int google::protobuf::MessageLite::ByteSize() const’ is deprecated: Please use ByteSizeLong() instead [-Werror=deprecated-declarations]
   42 |     unsigned ci_len = ci_msg.ByteSize(); \
      |                                       ^
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:529:3: note: in expansion of macro ‘SERIALIZE_AND_SEND_MSG’
  529 |   SERIALIZE_AND_SEND_MSG(func_name) \
      |   ^~~~~~~~~~~~~~~~~~~~~~
/home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:186:4: note: in expansion of macro ‘xclGetDebugMessages_RPC_CALL’
  186 |    xclGetDebugMessages_RPC_CALL(xclGetDebugMessages,ack,force,displayMsgs,logMsgs,stopMsgs);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/google/protobuf/generated_enum_util.h:36,
                 from /usr/local/include/google/protobuf/map.h:55,
                 from /usr/local/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/yasu/xilinx/XRT/build/Debug/runtime_src/core/pcie/emulation/hw_em/../common_em/rpc_messages.pb.h:26,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:25,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/usr/local/include/google/protobuf/message_lite.h:420:7: note: declared here
  420 |   int ByteSize() const { return internal::ToIntSize(ByteSizeLong()); }
      |       ^~~~~~~~
In file included from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:28,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:49:45: error: ‘int google::protobuf::MessageLite::ByteSize() const’ is deprecated: Please use ByteSizeLong() instead [-Werror=deprecated-declarations]
   49 |     _s_inst->sk_read(ri_buf,ri_msg.ByteSize()); \
      |                                             ^
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:529:3: note: in expansion of macro ‘SERIALIZE_AND_SEND_MSG’
  529 |   SERIALIZE_AND_SEND_MSG(func_name) \
      |   ^~~~~~~~~~~~~~~~~~~~~~
/home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:186:4: note: in expansion of macro ‘xclGetDebugMessages_RPC_CALL’
  186 |    xclGetDebugMessages_RPC_CALL(xclGetDebugMessages,ack,force,displayMsgs,logMsgs,stopMsgs);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/google/protobuf/generated_enum_util.h:36,
                 from /usr/local/include/google/protobuf/map.h:55,
                 from /usr/local/include/google/protobuf/generated_message_table_driven.h:34,
                 from /home/yasu/xilinx/XRT/build/Debug/runtime_src/core/pcie/emulation/hw_em/../common_em/rpc_messages.pb.h:26,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:25,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/usr/local/include/google/protobuf/message_lite.h:420:7: note: declared here
  420 |   int ByteSize() const { return internal::ToIntSize(ByteSizeLong()); }
      |       ^~~~~~~~
In file included from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h:28,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/home/yasu/xilinx/XRT/src/runtime_src/core/include/xcl_api_macros.h:50:55: error: ‘int google::protobuf::MessageLite::ByteSize() const’ is deprecated: Please use ByteSizeLong() instead [-Werror=deprecated-declarations]
   50 |     rv = ri_msg.ParseFromArray(ri_buf,ri_msg.ByteSize()); \
[... 12 lines cut ...]
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/debug.cxx:35:
/usr/local/include/google/protobuf/message_lite.h:420:7: note: declared here
  420 |   int ByteSize() const { return internal::ToIntSize(ByteSizeLong()); }
      |       ^~~~~~~~
[ 19%] Building CXX object runtime_src/core/common/CMakeFiles/xrt_coreutil.dir/api/xrt_profile.cpp.o
[ 19%] Building CXX object runtime_src/tools/xclbinutil/CMakeFiles/xclbintest.dir/SectionAIEMetadata.cxx.o
[ 20%] Building CXX object runtime_src/xocl/CMakeFiles/xocl.dir/api/clCreateContextFromType.cpp.o
[ 20%] Building CXX object runtime_src/tools/xclbinutil/CMakeFiles/xclbintest.dir/SectionBMC.cxx.o
[ 20%] Building CXX object runtime_src/tools/xclbinutil/CMakeFiles/xclbinutil.dir/SectionAIEMetadata.cxx.o
[ 20%] Building CXX object runtime_src/core/common/CMakeFiles/xrt_coreutil.dir/api/xrt_xclbin.cpp.o
[ 21%] Building CXX object runtime_src/core/common/CMakeFiles/xrt_coreutil.dir/config_reader.cpp.o
[ 21%] Building CXX object runtime_src/core/pcie/linux/CMakeFiles/core_pcielinux_objects.dir/device_linux.cpp.o
[ 21%] Building CXX object xma/src/xmaapi/CMakeFiles/xma2api.dir/xmascaler.cpp.o
In file included from /usr/include/boost/detail/endian.hpp:9,
                 from /home/yasu/xilinx/XRT/src/runtime_src/core/common/api/xrt_kernel.cpp:54:
/usr/include/boost/predef/detail/endian_compat.h:11:161: note: #pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
   11 | #pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")
      |                                                                                                                                                                 ^
[ 21%] Building CXX object xma/src/xmaapi/CMakeFiles/xma2api.dir/xmasignal.cpp.o
[ 21%] Building CXX object runtime_src/tools/xclbinutil/CMakeFiles/xclbinutil.dir/SectionBMC.cxx.o
cc1plus: all warnings being treated as errors
make[2]: *** [runtime_src/core/pcie/emulation/hw_em/CMakeFiles/hw_emu_objects.dir/build.make:89: runtime_src/core/pcie/emulation/hw_em/CMakeFiles/hw_emu_objects.dir/generic_pcie_hal2/debug.cxx.o] エラー 1
make[1]: *** [CMakeFiles/Makefile2:2530: runtime_src/core/pcie/emulation/hw_em/CMakeFiles/hw_emu_objects.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
```

uday610, 2020-10-11:
Maybe you can put more details, like which branch you are trying to build, which OS and versions? whether you have run xrtdeps.sh before the build, etc

pdisho, 2020-10-26:
I get the same error when trying to build 2019.2 XRT on Ubuntu 18.04. I ran xrtdeps.sh before the build and can't seem to find where I should disable the check for "[-Werror=deprecated-declarations]" or if that would be a bad idea. Any help would be appreciated.

pdisho, 2020-10-28:
I took out -Werror in src/runtime_src/CMakeLists.txt and it built successfully.

tinlans, 2020-10-30:
No matter which distribution is used, everyone could reproduce this issue with the latest protobuf library. My distribution is Gentoo Linux and the protobuf version is 3.13.0. Simply replace all ByteSize() calls with ByteSizeLong() could solve this error.

By the way, the pragma message related to BOOST_BYTE_ORDER will become an preprocessor error because the header file boost/detail/endian.hpp is removed from latest c++ boost libraries. However, it's unrelated to the topic of this issue.

keryell, 2021-03-10:
After few months of rest, I have just tried to compile the latest XRT and there is still this bug after 6 months and at least 4 folks reporting it... :-(

My modest contribution: https://github.com/Xilinx/XRT/pull/4895