Xilinx/XRT #4224: U50: runtime errors from HBM channels

boxerab, 2020-10-13:
I have four concurrent OpenCL kernels, and each kernel is linked to the HBM channels as follows:

```
sp=GainAdjustKernel_1.p_Input:HBM[0]
sp=GainAdjustKernel_1.p_Output:HBM[1]
sp=GainAdjustKernel_2.p_Input:HBM[2]
sp=GainAdjustKernel_2.p_Output:HBM[3]
sp=GainAdjustKernel_3.p_Input:HBM[4]
sp=GainAdjustKernel_3.p_Output:HBM[5]
sp=GainAdjustKernel_4.p_Input:HBM[6]
sp=GainAdjustKernel_4.p_Output:HBM[7]
```
However, I have no pragmas in the OpenCL kernel linking memory buffers to HBM channels.

I get these errors;

```
XRT build version: 2.7.766
Build hash: 19bc791a7d9b54ecc23644649c3ea2c2ea31821c
Build date: 2020-08-17 16:52:05
Git branch: 2020.1_PU1
PID: 6244
UID: 1000
[Tue 13 Oct 2020 05:08:45 PM GMT GMT]
HOST: aaron-TRX40-DESIGNARE
EXE: /home/aaron/src/openfx/HostSupport/examples/Linux-debug/hostDemo
[XRT] WARNING: Argument '7' of kernel 'GainAdjustKernel' is allocated in memory bank 'HBM[0]'; compute unit 'GainAdjustKernel_1' cannot be used with this argument and is ignored.
[XRT] ERROR: Buffer connected to memory '0' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            0 |
|              7 |            0 |
+-------------------------------+
[XRT] WARNING: Argument '6' of kernel 'GainAdjustKernel' is allocated in memory bank 'HBM[1]'; compute unit 'GainAdjustKernel_2' cannot be used with this argument and is ignored.
[XRT] ERROR: Buffer connected to memory '1' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            1 |
|              7 |           -1 |
+-------------------------------+
[XRT] WARNING: Argument '6' of kernel 'GainAdjustKernel' is allocated in memory bank 'HBM[2]'; compute unit 'GainAdjustKernel_3' cannot be used with this argument and is ignored.
[XRT] ERROR: Buffer connected to memory '2' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            2 |
|              7 |           -1 |
+-------------------------------+
[XRT] WARNING: Argument '6' of kernel 'GainAdjustKernel' is allocated in memory bank 'HBM[3]'; compute unit 'GainAdjustKernel_4' cannot be used with this argument and is ignored.
Enqueued and execute kernels @ 5.87254 ms
Plugin Render 5190.73 ms
[XRT] ERROR: Buffer connected to memory '3' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            3 |
|              7 |           -1 |
+-------------------------------+
[XRT] ERROR: Buffer connected to memory '1' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            1 |
|              7 |            1 |
+-------------------------------+
Error: clEnqueueMapBuffer return CL_MEM_OBJECT_ALLOCATION_FAILURE.
Error: mapDeviceToHost (CL_QUEUE_CONTEXT) returned CL_MEM_OBJECT_ALLOCATION_FAILURE.
[XRT] ERROR: Buffer connected to memory '2' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            2 |
|              7 |            2 |
+-------------------------------+
Error: clEnqueueMapBuffer return CL_MEM_OBJECT_ALLOCATION_FAILURE.
Error: mapDeviceToHost (CL_QUEUE_CONTEXT) returned CL_MEM_OBJECT_ALLOCATION_FAILURE.
[XRT] ERROR: Buffer connected to memory '3' cannot be used as argument to kernel 'GainAdjustKernel' because kernel has no compute units that support required connectivity.
+-------------------------------+
| GainAdjustKernel              |
|-------------------------------|
| argument index | memory index |
|              6 |            3 |
|              7 |            3 |
+-------------------------------+
Error: clEnqueueMapBuffer return CL_MEM_OBJECT_ALLOCATION_FAILURE.
```

zohourih, 2020-10-19:
Have you explicitly allocated your buffers in the appropriate HBM channel as described below?

boxerab, 2020-10-19:
Thanks, yes I have:

```
					in_bank_ext[i].flags = i | XCL_MEM_TOPOLOGY;
					in_bank_ext[i].obj   = NULL;
					in_bank_ext[i].param = 0;

					out_bank_ext[i].flags = i | XCL_MEM_TOPOLOGY;
					out_bank_ext[i].obj   = NULL;
					out_bank_ext[i].param = 0;

					client_flags =  CL_MEM_EXT_PTR_XILINX;
					inPtr = in_bank_ext + i;
					outPtr = out_bank_ext + i;
```

stsoe, 2020-10-23:
Hello. It looks like both in_bank and out_bank are allocated in same bank, but that is not the connectivity you are showing for the linking.

boxerab, 2020-12-08:
@stsoe right you are! thanks.

closed 2020-12-08 by boxerab